riscv/riscv-cheri #274: need separate extension for c.csc, c.cscsp, c.clc, c.clcsp for Zcm* compatiblity

tariqkurd-repo, 2024-06-03:
We need to be able to remove c.csc, c.cscsp, c.clc, c.clcsp so that Zcm* extensions can be implemented.

jrtc27, 2024-06-03:
I don’t think you want that

tariqkurd-repo, 2024-06-03:
how else can we suport Zcmt/Zcmp on a CHERI core? The encodings conflict and the code size savings of c.sc etc. will be less than having push/pop, table jump and other future risc-v code-size reduction extensions.

it messes up the toolchain though support Zcm* - is that the concern?

jrtc27, 2024-06-03:
I would be very surprised if not being able to compress loads and stores of pointers doesn't hurt you overall, and it would be asymmetric with loads and stores of integers. If you want Zcmt/Zcmp then something needs re-encoding so they don't clash.

arichardson, 2024-06-03:
I think we will need to do some analysis to decide which encodings are most beneficial. I would assume compressed load store are similarly beneficial to push pop - maybe we need to remove something else for c.csc. Possibly the sp-relative ones are less important with push-pop, but I'd still assume they are common.

tariqkurd-repo, 2024-06-03:
This one is key to get right for smaller RV32 embedded cores, if we can't get the code size reduction extensions then embedded-CHERI is going to look quite uncompetitive.

Maybe we do need to re-encode C.SC etc for RV32, something to think about. Maybe we should replace C.FLW etc. which are pretty much useless.

tariqkurd-repo, 2024-06-04:
I've been reviewing the encodings and had a look at https://github.com/riscv/riscv-zilsd
This is all for RV32 only

There are three groups of 16-bit encodings for load/store instructions.

`C.FLW`, replaced by `C.LD` by `Zilsd`, replaced by `C.LC` by `CHERI`
`C.FSD`, replaced by `Zcmp`, `Zcmt`, and future code-size reduction
`C.LW`, never replaced

(RV64 replaces `C.FLD` with `C.LC`, and there's no strong requirement for Zcm*)

So I think we need to be incompatible with `Zilsd`, choosing `C.LC` as preference, and not offer the option of compatibility.

So overall there's no change required, possibly except for noting the incompatibility with `Zilsd`. To support the D extension on RV32, and 16-bit encodings, and Zcm* the requirement is to specify `RV32DZcaZcmpZcmt` which is standard RISC-V.

tariqkurd-repo, 2024-06-04:
This is a confusing topic, and the text inthe current spec wasn't quite right.
I've tried to clarify the situation here: https://github.com/riscv/riscv-cheri/pull/280

tariqkurd-repo, 2024-06-05:
closed - so the conclusion is it all hangs together for RV32 - as the Zcmp/t instuctions don't conflict with C.LC etc.
For RV64 they _do_ conflict so we keep C.LC etc. Zcmp/t.

closed 2024-06-05 by tariqkurd-repo