riscv/riscv-cheri #283: The TID extension is insufficient for the current compartmentalisation models

francislaus, 2024-06-06:
The TID extension defines two registers: STID(C) and UTID(C) of which the latter is read-only. In order to support compartmentalisation models on multiple privilege levels running independently, I propose to add a read-write CSR on all support privilege levels resulting in: MTID(C), STID(C), and UTID(C).

Furthermore, I propose enforcing the following permissions on the CSRs:
MTID(C) is writeable with from M mode with ASR permission, but readable from M mode without ASR permission.
STID(C) is writeable with from M mode with ASR permission or S mode with ASR permission, but readable from M mode without ASR permission or S mode without ASR permission.
UTID(C) is writeable with from M mode with ASR permission or S mode with ASR permission or U mode with ASR permission, but readable from M mode without ASR permission or S mode without ASR permission or U mode without ASR permission.

I propose the following indexes for the CSRs:

`mtidc`: 0x780
`stidc`: 0x580 (unchanged)
`utidc`: 0x480

francislaus, 2024-06-06:
As a note, which one might add (or not add) to the spec: Having `xTID` capability registers adds the need to seal the content of these registers as it is otherwise unprotected from unprivileged code.

jrtc27, 2024-06-06:
> As a note, which one might add (or not add) to the spec: Having `xTID` capability registers adds the need to seal the content of these registers as it is otherwise unprotected from unprivileged code.

Not necessarily; only if the capability itself needs to be protected. It could just be a pointer to memory (that itself could contain sealed capabilities, but could just be a bunch of integers), though in that case you’d presumably want to make it a read-only capability.

PR #284 (2024-06-20): Extended tid chapter for m mode capabilities
This will close #283.
commit 912c9346 (2024-06-20): Extended tid chapter for m mode capabilities (#284)

This will close #283.
closed 2024-06-20 by tariqkurd-repo