riscv/riscv-cheri #286: Misaligned capability exception priority

PRugg-Cap, 2024-06-17:
The synchronous exception priority table (https://github.com/riscv/riscv-cheri/blob/2b1e45692c46d7baab357b5c3d65d3bc2cdfbe4a/src/riscv-integration.adoc?plain=1#L699-704) lists capability address misalign exceptions in the same box with the other CHERI faults. This is ambiguous in terms of priority, but also wrong because it implies the exception cause code should be 0x1c.

I imagine they should just be moved into the box below with the other alignment exceptions, but before "Optionally:". However, I didn't jump straight to writing the PR because I wanted to check we don't want to give them a different (CHERI) exception code, since the behaviour of the exception handler is potentially very different (m mode isn't allowed to emulate the misaligned access)?

jrtc27, 2024-06-17:
The trap handler needs to decode the instruction to figure out what you were doing anyway, so there isn't really much point having a new code for it. All it would do is let you fast-path the failure, but it's an exceptional error case so not something that needs any optimisation.

PRugg-Cap, 2024-06-17:
Okay, thought so... I was imagining a handler just shifting the width code, but such a handler would be broken anyway. I'll do the easy fix.

closed 2024-06-18 by arichardson