riscv/riscv-cheri #293: Representability check description doesn't capture address-wrapping edge cases

PRugg-Cap, 2024-06-21:
The description of the representability check is slightly imprecise in cases where the base or top want to wrap the address space. A correction is described in the bounds section: https://github.com/riscv/riscv-cheri/blob/8f587528c7d1a0005be68c38d955408bb22485f2/src/cap-description.adoc?plain=1#L501-L502

but not in the representable region section, where it is also relevant:
https://github.com/riscv/riscv-cheri/blob/8f587528c7d1a0005be68c38d955408bb22485f2/src/cap-description.adoc?plain=1#L649-L653 

In addition, we need to clarify what happens if the base wraps.

I'll need to think about how to describe this precisely, but just wanted to record the issue so it doesn't get missed. I don't think there's a bug, but somebody trying to implement it just from this description would certainly get confused!

tariqkurd-repo, 2024-07-01:
PR for this one? @PRugg-Cap ?

PRugg-Cap, 2024-07-01:
Not yet, sorry, I'll try tomorrow!

tariqkurd-repo, 2025-06-10:
fixed by the referenced MR

closed 2025-06-10 by tariqkurd-repo