riscv/riscv-cheri #295: SCSS & CBLD malformed arch perms

jamie-melling, 2024-06-25:
CBLD states that `cs2` permissions could have been produced by ACPERM, but, for completeness, this check should probably be expanded to `cs1` as well.
SCSS has no mention of how to handle arch perms that ACPERM could not produce, but should mimic the behaviour of CBLD in this regard and return 0 if either `cs1` or `cs2` has malformed permissions.

PRugg-Cap, 2024-06-25:
Agreed on both of those. While we're here, I think `SCBNDS` and friends are being too strict in checking the reserved bits: they should just check the bounds aren't malformed.

tariqkurd-repo, 2024-07-01:
https://github.com/riscv/riscv-cheri/pull/299 for the original request

why change the reserved bit behaviour? I was thinking that if a reserved bit is set then keeping the current behaviour means that the new extension has the option of changing the definition of all instructions defined that way.

Maybe this should be clearly stated on the individual instruction pages.

What exactly would you update @PRugg-Cap ?

PRugg-Cap, 2024-07-02:
@tariqkurd-repo Hmm, good point. It's inconsistent between `SCBNDS` and `ACPERM` though, whereas we probably expect that those bits are more likely to be used as permissions than extra bounds. Anyway, let's close this if the original issue is solved, and I can open a new issue if this seems significant enough to address.

tariqkurd-repo, 2024-07-03:
ok, closing for now

closed 2024-07-03 by tariqkurd-repo