riscv/riscv-cheri #298: Preserve tag if CBLD checks fail

PRugg-Cap, 2024-06-30:
https://github.com/riscv/riscv-cheri/issues/186 got fixed before I noticed, but I think we can safely fix it the other way, i.e. the "otherwise" case changes from "copy cs2 to cd and clear cd 's tag", to "copy cs2 to cd, leaving the tag unchanged". This is a behaviour we've referred to in the past as "CCBuildCap", i.e. conditional buildcap: try your best to tag the capability, but leave it as is if something goes wrong.

This could accelerate chains of buildcaps when you don't know which capability authorises the buildcap, e.g. if you want to restore tags on capabilities that could be from the stack or the heap, you can just buildcap by both in consecutive instructions without needing branches.

If software *really* wants to ensure the resulting cap is only tagged if it's a subset of the authorising cap, then it can clear the tag before running `CBLD`.

It wouldn't be the end of the world to leave things as is, but seems like a slight missed opportunity.

Note: As a side effect of this change, `CBLD` authorised by `cnull` would be safe to use as a `CMV`, so we could potentially get back an encoding (`ccleartag`?). However, that does complicate things slightly if an extension adds back in otypes, as then `CBLD` needs to clear the otype when setting the tag.

arichardson, 2024-07-01:
However, I'm not sure if we would want those semantics for superset unsealing (#199), since you probably do want to check that you got the expected input in that case. I can see use-cases for both semantics, not sure which one results in faster code in general.

tariqkurd-repo, 2025-06-10:
We still don't have a CCBuildCap equivalent - does it need to be in the first ratification proposal?
I'd prefer to tag this as `future enhancement` - do we need to handle it now?