riscv/riscv-cheri #302: No GCMODE

jrtc27, 2024-07-01:
We have SCMODE but no way to extract it back out. For RV64 one could slice it out of the high bits, but this gets rather ugly on RV32.

tariqkurd-repo, 2024-07-01:
it would be cheap to add it back in

tariqkurd-repo, 2024-07-02:
We also have this text: `the mode can be optionally switched using MODESW, and the result observed in dinfc.M.` in debug mode which is incorrect, which caused this discussion. The result _can't_ be observed without `GCMODE` as integer mode only reads the address field.

arichardson, 2024-07-02:
Reading the current mode can be done by executing something like `auipc; gctag/gchi` but reading the mode from arbitrary capabilities requires awkward bitmanip or a dedicated instruction.

tariqkurd-repo, 2024-07-02:
but `AUIPC` in integer mode doesn't return the metadata, it just gives in integer result.So maybe we can say that if you do `AUIPC` with zero offset and the tag isn't set then it must be integer mode, and if it is set then it's capability mode.

so `GCMODE` becomes `AUIPC x, 0; GCTAG y, x`

y=1 for cap mode, y=0 for int mode

Is that good enough?

PRugg-Cap, 2024-07-02:
I think that's what @arichardson meant, but the point is there's no way to know what the mode bit is for a cap that isn't in PCC, i.e. what the mode will be when you jump to it. I guess there's a question about how common that would be?

tariqkurd-repo, 2024-07-02:
`bext` can be used to extract it for RV64, trickier for RV32 because of the encoding.
Yes - it's all about whether it's common enough to care about it.

tariqkurd-repo, 2024-07-03:
In debug mode though AUIPC may not be valid - as PC relative instructions may be illegal.
Does the debugger care?
Should we add `GCMODE` for this reason? So the mode can be observed in debug mode?

jrtc27, 2024-07-03:
GCMODE will also be needed for instruction emulation trap handlers.

tariqkurd-repo, 2024-07-03:
How exactly? Can you give more details @jrtc27 ?

jrtc27, 2024-07-03:
You need to know the mode for xEPCC to know how to decode the instruction and figure out what to do.

tariqkurd-repo, 2024-07-03:
I guess you can do that by reading `xEPCC`, using using `BEXT` , and trickier decoding for RV32

But if `AUIPC` is illegal in debug mode, there's still no way of observing the mode in debug mode without `GCMODE`, which seems like a weak argument for a new encoding, maybe we can say something like the LSB of `dinfc` is the current mode, or something like that.

PRugg-Cap, 2024-07-03:
@tariqkurd-repo I think you can't do that even *with* `GCMODE`, unless you mean for `GCMODE` to get the *current* mode of `PCC`, rather than the mode of a capability in a register?

jrtc27, 2024-07-03:
I'm not talking about debug mode at all.

tariqkurd-repo, 2024-07-03:
Good point, so it doesn't solve that problem.

jrtc27, 2024-07-03:
(I'm talking about firmware, hypervisors or kernels needing to know what mode the software running on top of it is in, whether because it affects the SBI interface, or because it needs to decode the instruction that xEPCC refers to)

andresag01, 2024-07-03:
> (I'm talking about firmware, hypervisors or kernels needing to know what mode the software running on top of it is in, whether because it affects the SBI interface, or because it needs to decode the instruction that xEPCC refers to)

@jrtc27 : This is a good use-case. An instruction that simply gets the mode of an arbitrary capability in a register is helpful for this. However, please note that it will not tell you the _effective_ CHERI execution mode because that is a combination of CSRs and the PCC M-bit that the process was using at the time. Given that we provide such an instruction, is it reasonable to still expect software (e.g. Hypervisors, firmware, kernels, etc) to also look at the CSR state and work out the effective mode? This will probably take quite a few instructions (haven't worked it out yet!)

jrtc27, 2024-07-03:
Yes, and there's no good way to avoid that given the effective encoding mode also requires knowing what privilege mode you came from, so that would need to be an input to a hypothetical all-powerful instruction. The alternative would be to make it illegal to have M=0 whilst CHERI is disabled (though not as an illegal instruction exception, since you'd still need to disambiguate the cause there :)).

andresag01, 2024-07-03:
There could be an instruction that tells you the "effective" CHERI mode given the cap in xEPCC and other relevant CSRs for example although that sounds hard (and fiddly!) to implement in hardware...

Do you know in which cases the hypervisor, kernel, etc needs to know the effective CHERI mode of the other process? It would be great to have a concrete example use-case; perhaps there is something else we can do to improve this.

jrtc27, 2024-07-03:
The two that immediately come to mind are:

1. Misaligned access emulation in firmware (or soon the kernel, in the case of Linux, which wants to handle it itself)

2. MMIO device trap-and-emulate for hypervisors (UART, RTC, (A)PLIC, PCIe, VirtIO, ...)

Whilst the uncompressed instructions likely don't need to know the encoding mode, the compressed ones do due to repurposing encoding space.

tariqkurd-repo, 2024-07-04:
Uncompressed instructions do need to know though don't they? If you're emulating misaligned you need to know if the auth cap is DDC or cs1

The handler can always read CRE for the lower privilege level which it is emulating can't it? And use that when deciding how to decode.

tariqkurd-repo, 2024-07-04:
and none of this solves this problem:

`the current mode can always be observed in dinfc.M.`

which is not true in debug mode, so I should remove this from the spec.

tariqkurd-repo, 2024-07-04:
Jamie is suggesting that `GCMODE rd, x0` returns the M-bit from the current PCC in all cases, including debug mode which solves the problem of not observing the mode in debug mode.

jrtc27, 2024-07-04:
I mean it wouldn’t be GCMODE rd, c0 because that would mean precisely that. What you’re proposing is to make GCMODE’s cs1 have a constraint that it’s not c0 and use that spare encoding as GCMODE rd, pcc. That would be quite the exception to the proposed extension.

tariqkurd-repo, 2024-07-04:
yes - `GCMODE rd, cs0` would do `GCMODE rd, pcc`. It's irregular, but otherwise we have no way of reading the mode in debug mode.

jrtc27, 2024-07-04:
I mean, there should be a way to read the mode without requiring a program buffer.

tariqkurd-repo, 2024-07-04:
an alternative for debug mode would be to use one of the lower 64-bits of the `dinfc` CSR, such as bit 0, to be read-only representing the mode.

tariqkurd-repo, 2024-07-04:
This can work as follows:

`csrr cd, dinfc`
which will either read the whole thing (cap mode) or only the address field (int mode).
`cgettag` then will return 1 for cap mode or 0 for int mode

similarly for getting it from the PC, when `auipc` is used instead of `csrr`

I'll make a PR - and also note that the M-bit is _not_ directly observable in `dinfc` which is currently incorrectly stated in the doc

andresag01, 2024-07-08:
@tariqkurd-repo @jrtc27 : Did this discussion reach a conclusion? Do we need a new get-tag instruction or something else?

tariqkurd-repo, 2024-07-08:
Only for performance, not for functionality

arichardson, 2024-07-08:
We can have the compiler lower __builtin_cheri_mode_get() to the appropriate checks+bitslicing,
e.g. `gchi; bexti <m_pos>; bexti <x-pos>; and` (plus for rv32: `bexti <perms[4]>; xor` to check we are in the right permissions quadrant. Can probably be optimized a bit further as well using shift+andi. 4-6 instructions for something rather uncommon (trap handler emulation) should be fine IMO.

jrtc27, 2024-08-27:
That would get really weird with ACPERM.

arichardson, 2024-08-27:
> That would get really weird with ACPERM.

Agreed, I much prefer a dedicated instruction, but we don't get the requested opcode space, we could potentially change acperm to something like chgperm that behaves differently for bit 31. But I'd only do that if really needed.

jrtc27, 2024-08-27:
GC* are cheap when it comes to encoding space given they have a single source

davidchisnall, 2024-08-28:
They're also likely to be very rare, so could go in the 48-bit encoding space if anyone ever uses that.

jrtc27, 2024-08-28:
I have no objection to the increased code size from rare CHERI instructions using >32-bit encodings, but I do not think we want to be the first extension seriously using such encodings. Someone else can be the trailblazer for that...

davidchisnall, 2024-08-28:
> I do not think we want to be the first extension seriously using such encodings.

Coincidentally, that's what the authors of every other extension said.

tariqkurd-repo, 2024-08-28:
There's no ratified scheme for 48-bit encodings yet. There's a new proposal
but I haven't seen it yet.

PR #350 (2024-08-29): Add GCMODE instruction
Add GCMODE instruction that decodes the mode from an arbitrary capability in an *x* register and writes the value to another *x* register.

GCMODE allows inspecting a capability's CHERI execution mode without the need for software to "manually" decode the capability's metadata using `gchi`.

Fixes #302.
commit d63f4b56 (2024-08-29): Add GCMODE instruction (#350)

Add GCMODE instruction that decodes the mode from an arbitrary
capability in an *x* register and writes the value to another *x*
register.

GCMODE allows inspecting a capability's CHERI execution mode without the
need for software to "manually" decode the capability's metadata using
`gchi`.

Fixes #302.
closed 2024-08-29 by andresag01