riscv/riscv-cheri #303: M bit should be inverted

jrtc27, 2024-07-01:
Currently, we have M=1 for capability mode, but M=0 either means integer mode, if supported, or capability mode, if integer mode is not. Moreover, the infinite capability is defined to have M=0, so you have to SCMODE it if you want to derive things from it that should remain in capability mode. It would be much more natural to have M=0 mean capability mode, as the thing that you are likely using if interacting with capabilities, and then the bit only needs to be encoded to represent a non-zero state when you want to use legacy mode, making things unambiguous. This also means the infinite capability would be capability mode. Note that null is unaffected since it is non-X and thus M does not exist there.

tariqkurd-repo, 2024-07-01:
why is the infinite-cap in integer mode?

I support this change - I think it makes the definition clearer if ti doesn't cause problems.

arichardson, 2024-07-01:
I guess this means we have to define the PCC reset capability to be != infinite-cap since it has to be integer mode if the core supports hybrid? That's fine by me but probably needs calling out explicitly?

tariqkurd-repo, 2024-07-02:
For 100% compatibility then it must boot into integer mode, yes.
The mode is orthogonal to the other features of the infinite-cap so I think we can make it work..... I'll give it a go

closed 2024-07-03 by arichardson