riscv/riscv-cheri #313: Section 5.7/8.5 clarification

buxtonpaul, 2024-07-05:
In section 5.7 Disabling CHERI Registers the description points to Sections 3.3 and 8.5 for instructions that should generate exceptions. Whilst the text does say CHERI instructions, it is quite easy to read it as all instructions in table 8.5 would generate an exception.
Suggest either
1. Add column to the table in 8.5 to indicate clearly the instructions specified.
2. Reword the sentence to make it clear that not all the instructions in 8.5 are CHERI instructions. E.g.
The instructions in Section 3.3 and the CHERI instructions in Section 8.5 cause illegal instruction exceptions.

tariqkurd-repo, 2024-07-15:
it's certainly not very clear, it should be the instructions _added_ (not _modified_) by the addition of Zcheripurecap and Zcherihybrid. I'm not sure such a list exists in the spec, as they often include the _modified_ instructions such as in table 31 (section 8.3)

tomaird, 2024-07-24:
Additionally, should the instructions added by `Zish4add` and `Zabhlrsc` also generate exceptions? One would assume not since those extensions aren't mentioned, but then again they are added by the same spec so it might be worth a clarification too to make it explicit.

tariqkurd-repo, 2025-06-10:
In the unpriv spec draft `Zish4add` and `Zabhlrsc` are separate extensions from CHERI and so not subject to CRE.
It says `Instructions that access c registers (implicitly or explicitly) cause illegal instruction exceptions`, and notes that CRAM is excluded.

closed 2025-06-10 by tariqkurd-repo