riscv/riscv-cheri #322: Integer pointer mode in debug mode

andresag01, 2024-07-10:
The current CHERI RISC-V spec goes to great length to provide a "compatibility" mode with CHERI-less RISC-V when executing code in M, S and U mode provided that Zcherihybrid is implemented. However, the machine always enters debug in capability pointer mode. Should there be an option to enter debug mode in "integer pointer mode", perhaps something like a CRE?

NOTE: I am only pointing out that a CHERI machine supporting Zcherihybrid would need a CHERI-aware debugger even though, for whatever reason, it is only ever running non-CHERI aware code compiled by a non-CHERI aware compiler.

andresag01, 2024-07-12:
@tariqkurd-repo @jrtc27 : Do you have a view on this?

tariqkurd-repo, 2024-07-15:
The debugger only needs to issue an SCMODE before running existing debug code, but it's a reasonable point that something does have to change. I think it's normal to have to customise debugger software for specific cores anyway, just look at how many debug options there are in the RISC-V spec. Also I think that a CHERI core is likely to be delivered with a customed debugger, which in this case will include the update to switch modes, or more likely to have full CHERI support available in the debugger.

Your proposal is simple enough - but I'd hold off until the need for it is demonstrated.

I'm pretty sure that @jrtc27 will have an opinion.

tariqkurd-repo, 2024-09-12:
this has been raised as a real problem, as entering capability mode in debug mode by default prevents a legacy debugger from functioning.

jrtc27, 2024-09-12:
I don't see why that matters. If the debug spec ever bumps its version again then you'll need a debugger that understands that. It already specifies 3 versions, 2 of which are widely used, and I doubt any debugger out there supports RV128 which would similarly be broken the moment you ship 128-bit hardware even if it also supports 64-bit.

tariqkurd-repo, 2024-09-17:
I agree - and I also know that debuggers are very expensive - and getting them customised makes them even more expensive - so having a "risc-v compliant" debug interface is really important, whatever that means with the various options in the debug spec.

tariqkurd-repo, 2024-09-17:
https://github.com/riscv/riscv-cheri/pull/370

closed 2024-09-23 by tariqkurd-repo