riscv/riscv-cheri #323: ACPERM should clear the tag if reserved bits are set

PRugg-Cap, 2024-07-10:
(This is the follow up I mentioned in https://github.com/riscv/riscv-cheri/issues/295)

We currently clear the tag if we try to do an SCBNDS on a cap with non-zero reserved bits. For consistency, we should do the same on an ACPERM.

Presumably the reason for the SCBNDS behaviour is to pave the way for those extra bits being used to change the bounds interpretation, while still being compatible. Since the bits are more likely to be used for permissions than for bounds, it seems odd to not have the same defence.

(Here's hoping this is the last malformed-related tweak!)

andresag01, 2024-07-11:
@tariqkurd-repo : Are you planning to create a PR for this one? Thanks!

PRugg-Cap, 2024-07-11:
(As said in the meeting, I'm happy to make the PR if we agree it should be done: it shouldn't be too tricky)

tariqkurd-repo, 2024-07-15:
This sounds ok to me - @PRugg-Cap are you able to make the PR (if you haven't already)?

closed 2024-07-16 by tariqkurd-repo