riscv/riscv-cheri #324: Allowing using alternative capability encodings

andresag01, 2024-07-12:
As discussed in this week's CHERI TG meeting, it is sometimes desirable to use alternative capability encodings, particularly in RV32. However, the CHERI RISC-V specification currently mandates capability encodings for RV32 and RV64 in the base extension Zcheripurecap. A proposal to get around this restriction is to split out the capability encoding (i.e. most of Chapter 3) into a separate CHERI RISC-V extension and include a paragraph indicating that the encoding described in this specification could be replaced with an alternative encoding.

There is precedent for a similar arrangement in the ratified RISC-V privileged specification which has the following text as a note in the introduction:

> We briefly note that the entire privileged-level design described in this document could be replaced with an entirely different privileged-level design without changing the unprivileged ISA, and possibly without even changing the ABI...... For simplicity of expression, the text is written as if this was the only possible privileged architecture.

The CHERI RISC-V spec would require that alternative encodings at the very least support the minimal set of features that underpin the base CHERI instructions. For example, the alternative encoding would need to support all the base architectural permissions and sentries, and could add new permissions.

To enable this flexible arrangement, we would need to review the existing CHERI RISC-V instructions and features to ensure that nothing is defined in such a way that it is directly dependent on the capability encoding. For example, the current description of ACPERM springs to mind because it restricts the allowed combinations of set permissions depending on what can be encoded in the AP field. We probably need to move these restrictions into the new capability encoding extension. Another example is probably the CHERI execution mode which is alluded to in Chapter 3 although this is not part of Zcheripurecap.

andresag01, 2024-07-12:
@davidchisnall, @tariqkurd-repo: What's your view on this?

davidchisnall, 2024-07-12:
I think it is the right approach.  I believe @nwf-msr has a concrete proposal along these lines underway.

andresag01, 2024-07-12:
Thanks! @nwf-msr could you share the details of your proposal?

nwf-msr, 2024-07-12:
Absolutely.  I have just posted it to the list: https://lists.riscv.org/g/tech-tg-cheri/message/6 .

PR #357 (2024-09-10): Allow alternative capability encodings
Allow implementations supporting CHERI RISC-V to use alternative capability encodings that provide most key features, such as bounds and permissions, but may change, for example, the granularity of bounds or offer additional features. Also, allow the `sentry` instruction to become an illegal when using alternative capability encodings that do not provide ambient sealing permission.

The paragraph allowing alternative encodings is largely inspired from the paragraph in the RISC-V Privileged Specification that allows alternative privileged ISA designs.

Fix [... cut]
commit cf788fdd (2024-09-10): Allow alternative capability encodings (#357)

Allow implementations supporting CHERI RISC-V to use alternative
capability encodings that provide most key features, such as bounds and
permissions, but may change, for example, the granularity of bounds or
offer additional features. Also, allow the `sentry` instruction to
become an illegal when using alternative capability encodings that do
not provide ambient sealing permission.

The paragraph allowing alternative encodings is largely inspired from
the paragraph in the RISC-V Privileged Specification that allows
alternative privileged ISA desig [... cut]
closed 2024-09-10 by andresag01