riscv/riscv-cheri #331: Does senvcfg.CRE retain its value?

Timmmm, 2024-07-18:
The spec says `senvcfg.CRE` is read-only zero in various dynamic situations (e.g. if `menvcfg.CRE` is zero). However it does say if the underlying value is retained and just *reads* as zero, or if the values is actually mutated to be zero.

I.e. what does this yield?

```
menvcfg.CRE = 1
senvcfg.CRE = 1
menvcfg.CRE = 0
menvcfg.CRE = 1
print(senvcfg.CRE)
```

buxtonpaul, 2024-07-18:
My reading of the spec has been that it would print 1.

Timmmm, 2024-07-19:
Yeah I think that is the most logical thing. There's a bit in `pmpaddr` that behaves similarly:

>  Although changing `pmpcfg.A[1]` affects the value read from `pmpaddr`, it does not affect the underlying value stored in that register—in particular, `pmpaddr[G-1]` retains its original value when `pmpcfg.A` is changed from NAPOT to TOR/OFF then back to NAPOT.

 What bit of the spec are you basing that on though?

Timmmm, 2024-07-19:
I guess there are actually three options:

1. Setting `menvcfg.CRE` to 0 clears `senvcfg.CRE` (bad option IMO).
2. When `menvcfg.CRE` is 0, `senvcfg.CRE` *reads* as 0, but is still writable.
3. When `menvcfg.CRE` is 0, `senvcfg.CRE` *reads* as 0, and is read-only.

I think option 2 would be easiest to deal with and matches `pmpaddr`.

buxtonpaul, 2024-07-19:
More a lack of an explicit statement that it would clear the bit, and the expectation that if the menv bit changes for some reason and then changes back, the s-mode would not be expected to be aware of this and would expect to be able to continue with the behaviour it expects.

jrtc27, 2024-07-19:
SSE deviates from that pattern, FWIW. There's an argument that it's confusing to be able to turn on CRE and observe it as 1 yet it not work because of a more-privileged mode setting it to 0, but that's true of CBCFE and CBZE, and you're supposed to just not expect them to do anything without first verifying you have the relevant extensions, e.g. via the ISA string in the FDT.

Timmmm, 2024-07-23:
What's SSE? My google skills are failing me...

Yeah I agree it's slightly awkward for software... but the worst thing is they have to check 3 CSRs and AND the bits which doesn't seem too bad. Arguably the fact that the CSR value might be zeroed without them actually writing it is worse. E.g if an OS sets `senvcfg.CRE` it could randomly be zeroed by the firmware toggling `menvcfg.CRE`... does the OS have to keep double checking `senvcfg.CRE` and setting it back to 1 or something? (I may be way off there - I'm not too sure about the use cases of this bit.)

Either way, avoiding this unspecified behaviour and getting rid of a WARL field (a cross-CSR WARL field even!) and seems easily worth it to me.

arichardson, 2024-07-23:
If I recall correctly we specified it this way since otherwise there is no way of determining the effective CRE for e.g. S-mode, since menvcfg is read-only. But maybe it's fine to just require m-mode to pass on this information or have s-mode probe whether CHERI instructions/CSRs trap.

I'm fine with the simpler behaviour and as Jess said this kind of behaviour already exists for other RISC-V extensions.

Timmmm, 2024-07-23:
> there is no way of determining the effective CRE for e.g. S-mode, since menvcfg is read-only

Don't you only need read access? Maybe I've misunderstood something...

Timmmm, 2024-07-23:
Ah good point, I forgot about that. That does make it slightly more awkward, but I guess that's true for loads of stuff already that is only visible in M-mode CSRs?

> What's SSE

Never mind, I found it. Somewhat predictably the spec for SSE also doesn't specify whether `senvcfg.SSE` retains its underlying value when `menvcfg.SSE` is 0.

PR #332 (2024-07-30): Make CRE not a WARL field
Change the CRE bits so they are always writable, but the effective CRE depends on multiple values. This is significantly simpler to specify, avoids an entire cross-CSR WARL field (WARL fields are a pain), and removes some undefined behaviour (do the bits retain their values when they are read-only zero).

Fixes #331

Note, I deleted the text about SXLEN/UXLEN. Those didn't seem fully specified tbh, but I'll need to add them back somehow.
commit 8b1b3b90 (2024-07-30): Make CRE not a WARL field (#332)

Change the CRE bits so they are always writable, but the effective CRE
depends on multiple values. This is significantly simpler to specify,
avoids an entire cross-CSR WARL field (WARL fields are a pain), and
removes some undefined behaviour (do the bits retain their values when
they are read-only zero).

Fixes #331

Note, I deleted the text about SXLEN/UXLEN. Those didn't seem fully
specified tbh, but I'll need to add them back somehow.

---------
closed 2024-07-30 by andresag01