riscv/riscv-cheri #333: Should the Zstid CSRs be guarded by CRE

tomaird, 2024-07-24:
The description for `CRE` says:

> Executing CSR instructions accessing any CSR added by Zcherihybrid (see Table 25) causes an illegal instruction exception

But this doesn't include the CSRs added by Zstid. Should they be included as well?

From @PRugg-Cap :

> We need to make sure sneaky user-mode software can't stash values that are hidden from legacy OSs. That means either TID needs to be guarded on CRE, or more likely (but more annoyingly) we need to have _another_ enable bit for the TIDs.

tariqkurd-repo, 2024-08-21:
they definitely need an enable bit - so sharing CRE seems most sensible.
opinions?

jrtc27, 2024-08-21:
If it’s a separate extension it needs its own enable. Even if it’s dependent on CHERI you could write software for CHERI without xtid, which will then be broken if you add xtid without its own enable.

PRugg-Cap, 2024-08-21:
I was worried you might say that...

arichardson, 2024-08-22:
I also think there this should not be gated by CRE, but instead using a state-enable bit. The capability extended versions should definitely be guarded by CRE though.

tariqkurd-repo, 2024-08-23:
yeah - a state enable seems likely to be required

closed 2024-09-10 by arichardson