riscv/riscv-cheri #334: Rename CHERI Register Enable (CRE)

Timmmm, 2024-07-25:
It's not a great name because it really gates access to all CHERI features - not just registers; you can't use instructions either.

Difficult to think of a really good name - the obvious `CHERI Enable` makes it sound like it would disable `ddc` checking which isn't the case. Best I could think of was `CA` - `CHERI Access`. Or `CIRE` - `CHERI Instruction and Register Enable`.

jrtc27, 2024-07-25:
Why should it gate instructions that don't touch CHERI state? In particular, CRRL and CRAM are safe to always be enabled. That's why it's CHERI Register Enable, not CHERI Enable; it gates access to the *state*, i.e. registers.

andresag01, 2024-07-25:
> Why should it gate instructions that don't touch CHERI state? In particular, CRRL and CRAM are safe to always be enabled. That's why it's CHERI Register Enable, not CHERI Enable; it gates access to the state, i.e. registers.

There are really 3 CHERI-related execution modes from what I recall:

1. Capability pointer mode
    * Everything is caps (like instruction operands, etc) in the usual way
    * Some instructions remapped for CHERI e.g. c.fsw becomes c.sc
2. Integer pointer mode
    * Most instruction operands are integers
    * Implicit use of DDC/PCC as authorising capability
    * Instructions are not re-mapped for CHERI -- they remain the same as regular RISC-V
    * We keep most other CHERI instructions like cram, scaddr, sentry, etc
3. Effective CRE=0
    * All instruction operands are integers
    * No instructions are re-mapped
    * All other CHERI instructions throw illegal instruction exceptions
    * All accesses to CHERI CSRs throw illegal exception
    * Implicit use of DDC/PCC as authorising capability
    
This arrangement has been in CHERI RISC-V for a few months now. The idea was to provide increasing degrees of backwards compatibility with regular, CHERI-less RISC-V to the point were you get almost a normal RISC-V machine in (3) with no extra instructions, CSRs, etc -- CRE=0 only keeps PCC/DDC checks which you can pretty much wave away when these are set to almighty cap.

The motivation for (3) was to (hopefully!) facilitate ratification by claiming that we can (pretty much) switch off CHERI and just get RISC-V.

andresag01, 2024-07-25:
We had a discussion about this today:
* We will not change the naming for now because it is likely to change again when the spec is reviewed for ratification
* Keep the effects of CRE bits as-is, but we will include a note about CRAM not changing CHERI state
* Add text to the spec clarifying the three CHERI-related execution modes

PR #396 (2024-10-03): Add table and notes about CRE
Fixes #334

* Add a table summarizing the hart's behavior in connection with the effective CRE and the CHERI execution mode
    * There are 3 cases CHERI disabled, Integer pointer mode and Capability pointer mode
* Add a note describing why CRAM is disabled when CRE=0
commit 01797dac (2024-10-03): Add table and notes about CRE (#396)

Fixes #334

* Add a table summarizing the hart's behavior in connection with the
effective CRE and the CHERI execution mode
* There are 3 cases CHERI disabled, Integer pointer mode and Capability
pointer mode
* Add a note describing why CRAM is disabled when CRE=0
closed 2024-10-03 by tariqkurd-repo