riscv/riscv-cheri #342: ACPERM description has inverted M-bit text

tomaird, 2024-08-14:
This section of the ACPERM description hasn't been updated since the meaning of the M-bit has been flipped:

[image]

I believe it should say:

```
3. M-bit cannot be 0 without X-permission being set
  a. Set M-bit to 1 unless X-permission is set

The combination of X-permission clear and M-bit clear is reserved for future extensions.
```

jrtc27, 2024-08-14:
No, the bit is always 0 for X=0 because it has no meaning then.

jrtc27, 2024-08-14:
For example see “Field values of the NULL capability”, which clearly shows M as 0 despite X also being 0.

tomaird, 2024-08-14:
Ok maybe M should still be 0 if X=0, but surely the first line in my screenshot is still wrong? 

"M-bit cannot be set without X-permission being set" - this now means you cannot set the mode to integer mode without X=1. Previously this meant you couldn't set the mode to capability mode without X=1.

jrtc27, 2024-08-14:
> Ok maybe M should still be 0 if X=0, but surely the first line in my screenshot is still wrong?

No.

> "M-bit cannot be set without X-permission being set"

Yes.

> - this now means you cannot set the mode to integer mode without X=1. Previously this meant you couldn't set the mode to capability mode without X=1.

There are no "integer mode" and "capability mode" for non-executable capabilities. The bit in the decoded permissions bit vector reads as 0 because it has to read as _something_, but it doesn't mean anything, and 0 is the normal value to use when something doesn't exist.

tomaird, 2024-08-14:
Ok I'd misunderstood. Thanks for clarifying.

closed 2024-08-14 by tomaird