riscv/riscv-cheri #344: Does the mepcc alignment legalisation use scaddr?

tomaird, 2024-08-19:
The spec describes how mepcc[0] (and sometimes mepcc[1]) are implicitly zeroed, and that its tag should be cleared if the legalised address is outside the representable range. However it doesn't mentioned if this is done using `scaddr` as is done for "invalid address conversion" (which is a separate mechanism).

It does say how the tag is _not_ cleared when reading/writing a sealed capability in `mepcc` if the original address matches the legalised address. But doesn't explicitly say that it _should_ be cleared when reading/writing a sealed capability in `mepcc` if the addresses do _not_ match. 

There are also other nuances to using `scaddr` to update the address, e.g. if the reserved bits are not zero.

This is tangentially related to #343, and the result may also apply to `dpcc` depending on the outcome of that issue.

andresag01, 2024-08-19:
This table has the information that you are looking for.
[image]

tomaird, 2024-08-19:
Yes I had seen this table, but my point is that it _doesn't_ say to use `scaddr` for CLEN writes to `mepcc` if the address changed. Should it?

PRugg-Cap, 2024-08-19:
Historically, clearing the LSB of `mepcc` couldn't make it unrepresentable, because the base was always well aligned. That's not true anymore with our fix for the AUIPCC representability problem. However, since the spec supports C being dynamically turned off and on, it needs to be on *read* and not on *write* that the legalisation happens. That means the write really is just the direct write. In terms of reserved bits, we've already said the tag gets cleared if we try to write a reserved/malformed cap, so shouldn't be an issue. As you say, if you're doing the invalid address conversion, that follows the usual rules.

andresag01, 2024-08-19:
@PRugg-Cap : Thanks for clarifying. The spec is missing a few bits then.

> However, since the spec supports C being dynamically turned off and on, it needs to be on read and not on write that the legalisation happens. That means the write really is just the direct write.

The representability check on read is a little inconvenient from the implementation's point of view. However, I think this is only needed in the case that the machine implements RV32 _and_ the C extension can be dynamically turned on/off. Right? I am going by this bit of text from the RISC-V privileged spec:

> If an implementation allows IALIGN to be either 16 or 32 (by changing CSR misa, for example), then,
whenever IALIGN=32, bit mepc[1] is masked on reads so that it appears to be 0. This masking occurs
also for the implicit read by the MRET instruction. Though masked, mepc[1] remains writable when IALIGN=32.

andresag01, 2024-08-19:
Yes, sorry, the RV32 is not relevant, its just the dynamic C switching

tomaird, 2024-08-19:
> That sounds right. We're building up a long list of caveats!

@PRugg-Cap - in that case what's the point of this note under `mepcc`:

> When reading or writing a sealed capability in mepcc, the tag is not cleared if the original address equals the legalized address.

This note only makes sense if you DO clear the tag for a sealed cap where the addresses differ, but the spec doesn't actually say that bit anywhere. And your comment confirmed that you shouldn't clear the tag in that case.

PRugg-Cap, 2024-08-19:
> AND is sealed

Oh apologies, I somehow missed this. No, the tag certainly needs to be cleared if the cap is sealed and gets changed by the read.

tomaird, 2024-08-19:
Ok thanks, that makes sense. But that definitely could be clearer from the wording in the spec!

tariqkurd-repo, 2024-08-21:
@PRugg-Cap will you make a PR to address this?

closed 2024-08-22 by arichardson