riscv/riscv-cheri #352: Clarify writing non-extended CLEN CSRs in integer pointer mode

tomaird, 2024-09-04:
In section 5.4.4 CSR Instructions, there is this section:

> Accessing a capability CSR other than an extended CSR in Integer Pointer Mode results in an illegal instruction exception. These CSRs are listed in Table 25.

Where Table 25 lists: dddc, mtdc, stdc, vstdc, ddc

Is it correct for these CSRs to raise an exception? This seems inconsistent with accessing other CLEN CSRs in integer pointer mode. If it is correct, what is the motivation behind it?

PeterRugg, 2024-09-04:
Hmm, that does seem odd. I suspect it's a hangover from before we had mode-dependent CSR access?

arichardson, 2024-09-04:
I think this might be from when we added new CSRs instead of extending existing ones.
Maybe the rationale was that updating the address on these is not useful behaviour and most likely to be a programming mistake.

I think this restriction is not required so for consistency we should probably drop it.

tomaird, 2024-09-04:
If this restriction is dropped, there should be some other clarification on what action to take on writing a non-extended CLEN CSR in integer pointer mode, ideally just adding them to the CLEN CSR tables at the end of the spec

tariqkurd-repo, 2024-09-09:
ok, this gets added to the list. I think the intended spec is clear, we just need to bring the doc up to date to match it.