riscv/riscv-cheri #362: Include SAIL snippets into the specification

andresag01, 2024-09-12:
This includes the snippets into the instruction listings as well as the capability description chapter.

francislaus, 2024-12-12:
Can we (and should we) pick up a dependency to the Sail model for that? Would it be possible to automatically pull in the relevant lines from the Sail model? @Timmmm @PeterRugg

wmat, 2024-12-12:
There is for PDFs.  See https://github.com/Alasdair/asciidoctor-sail

I have a branch of riscv-isa-manual called sail-inclusion here:
https://github.com/riscv/riscv-isa-manual/tree/sail-inclusion with an
example of using Alasdair's extension to pull SAIL into the doc.  If you
look at my commit here:
https://github.com/riscv/riscv-isa-manual/commit/abd45a5a56154bcabbd07daaf41d91a1945729ce

you can see the changes I had to make for the example.  But please see
Alasdair's documentation in his repository for more information here:
https://github.com/Alasdair/asciidoctor-sail/blob/master/doc/built/sail_to_asciidoc.pdf

Timmmm, 2024-12-13:
I did set this up when the spec was in Codasip but we removed it when it was moved to GitHub since the Sail code wasn't open at the time. I'll see if I can resurrect it.

The big question is how do you get the JSON file. It's very big so I don't think you just want to add it to Git. But adding Sail compilation to the build process is also not great (although it's a lot better since we made the Sail binary release). In Codasip we had a cache on a web server which worked pretty well. How do people feel about using GitHub releases for that?

So every push (or maybe tag) in the Sail repo would generate the JSON and add it to a GitHub release, and then this repo would just point to the URL and download it.

We'll have to answer this question for riscv-isa-manual too and I think this repo could serve as a good demonstrator.

wmat, 2024-12-16:
For my testing purposes on riscv-isa-manual I've been generating the index at the top of tree of the sail repository and adding it directly into the repo and referencing it with an AsciiDoc attribute in the document header (riscv-unprivileged.adoc) with `:sail-doc: src/sail-index/riscv_RV64.json`.

Timmmm, 2024-12-16:
Yeah the issue is it's 4.5 MB (3.1 MB minified). Feels a bit much for a file that we'd continually update? Though I guess it probably does diff pretty well at least.

Timmmm, 2024-12-19:
Ok I have update the Sail repo to have [a Github workflow to generate the JSON](https://github.com/CHERI-Alliance/sail-cheri-riscv/actions/runs/12417940452) and I have made [a release of it](https://github.com/CHERI-Alliance/sail-cheri-riscv/releases/tag/2024-12-19) (making a release is a manual process currently but I can automate it in future).

Next step is to update this repo to `curl https://github.com/CHERI-Alliance/sail-cheri-riscv/releases/download/2024-12-19/riscv_RV64.json` and add the `asciidoctor-sail` stuff. I'll have a go tomorrow.

Timmmm, 2024-12-20:
PR here: #498 
(on top of #497)

andresag01, 2025-01-29:
Hi @Timmmm , thanks for the contribution. The snippets look good! Would it be feasible after this change to include the SAIL snippets implementing the capability encoding? I am specifically referring to replacing [this](https://github.com/riscv/riscv-cheri/blob/main/src/cap-description.adoc?plain=1#L424)  pseudocode with SAIL -- that pseudocode is not particularly well-defined and it was always written with the intention to replace it with the actual SAIL.

Timmmm, 2025-01-30:
We could replace it with [this code](https://github.com/CHERI-Alliance/sail-cheri-riscv/blob/1a086b68c3ec6ef5ee10ec0b3ab8a5d385091093/src/cheri_cap_common.sail#L214-L247)... I don't think there's a way to intersperse it with prose though. We should probably just clean up the comments in the Sail code a bit.