riscv/riscv-cheri #375: xtval cannot reasonably be read-only zero for CHERI exceptions

jrtc27, 2024-09-20:
Surely we don't want to permit that, just as it's not permitted for, say, page faults?

andresag01, 2024-09-20:
I am guessing this is in reference to the RISC-V Privileged spec which allows implementations to not support `mtval`. I think this RISC-V "feature" is mostly for very tiny implementations that, for example, would only support M-mode for whatever reason.

I agree that it is not ideal to have an implementation that does not support xtval, but do we need to explicitly prevent that configuration in the CHERI RISC-V spec even though the RISC-V Privileged spec explicitly calls that out as an option?

tariqkurd-repo, 2024-09-26:
I'm inclined to agree with @jrtc27 , however I left it in there because I've had some visibility of really tiny cores where every gate counts. Maybe you'd want to have a really minimal implementation where every CHERI fault is simply fatal and the system is reset, without worrying about the exact fault time. I guess this kind of system is why mtval can be zero in the first place.

So I really agree that in almost all cases you'd want the error information, but I guess we shouldn't exclude  people making a tiny system where they don't have to have it.

closed 2024-09-26 by andresag01