riscv/riscv-cheri #389: Alternative semantics for AUIPC in Capability Mode

andresag01, 2024-10-01:
CHERIoT uses a different immediate width for AUIPC. We should add a note in the instruction listing for AUIPC indicating that extensions on top of CHERI RISC-V may re-map this instruction when the CHERI Execution Mode is Capability.

jrtc27, 2024-10-01:
This one sounds pretty dodgy to allow

nwf, 2024-10-02:
Recall discussion from https://github.com/riscv/riscv-cheri/issues/56 .

CHERIoT's capability encoding does not tolerate cursors going out-of-bounds as far as the Zcheripurecap proposed encoding, and so we found that we needed to use a smaller (by one) shift of our AUIPCC instruction.  I'd tried to raise this divergence in https://lists.riscv.org/g/tech-tg-cheri/message/6, but I think it got lost among everything involved.  I _think_ the only thing CHERIoT needs is for AUIPC's shift to be a function of the capability encoding; we surely don't need to replace the entire instruction arbitrarily.