riscv/riscv-cheri #402: SCBNDSI redundancy?

nwf, 2024-10-02:
`SCBNDSI` has a 5-bit immediate, `uimm`, and a gated (by `s`) shift-by-4 operation, in
https://github.com/riscv/riscv-cheri/blob/1c43ce2fe5688057d6108a5f901574f2dac0acd0/src/insns/scbnds_32bit.adoc?plain=1#L43 which means that there are two ways to request bound lengths of `0` (unshifed or shifted 0) and `16` (unshifed 16, shifted 1).  Perhaps "shifted 0" and "shifted 1" could be put to better use (as requests for `512` and `528`, say)?

arichardson, 2024-10-10:
Avoiding redundant encodings here would be nice and I agree that having shifted zero be `512` would make a lot of sense here. Not sure how much that increases decoder complexity, but if we are already special casing, maybe we could also have shifted 1 be something more useful like `1024` or `4096`?

davidchisnall, 2024-10-10:
I'd like to see some distribution of object sizes before thinking about adding that.  And it will probably be different on 32- and 64-bit systems.  It might be worth marking those as explicitly invalid encodings that are reserved for future expansion.

arichardson, 2024-10-10:
> I'd like to see some distribution of object sizes before thinking about adding that. And it will probably be different on 32- and 64-bit systems. It might be worth marking those as explicitly invalid encodings that are reserved for future expansion.

That is a good argument - I believe @jonwoodruff did some analysis here.
For now I've opened a PR to reserve these encodings.

jonwoodruff, 2024-10-11:
These are data samples from Spec2006, averaging distributions from each of 10 benchmarks.

This first set is for flat "bits of precision". It starts at 1 bit because we assume signed values.  12 reaches 100% because we're just measuring the used sizes in RISC-V code generation from CHERI ISA-v9, which only has a 12-bit immediate.

 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
incoffseti | 0.14% | 41.46% | 47.54% | 53.55% | 62.63% | 69.90% | 72.67% | 76.13% | 87.26% | 97.36% | 98.02% | 100.00%

And here are results with a "scale bit" that shifts by 4, which was the point that hit the highest number of cases.  Presumably this is because it facilitates pointer-aligned arithmetic. Unsurprisingly, this matches choices in the Morello ISA.

   | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
incoffseti | 0.14% | 40.06% | 49.83% | 66.03% | 72.97% | 85.50% | 96.57% | 97.22% | 98.09% | 98.29% | 98.53% | 100.00%

I don't know if this informs how effective adding an additional value, e.g. 512, to the spectrum.  I can dig up the old scripts and add that condition in. There is at most 10% on the table (the jump from 9 bits to 10 bits; we would get a single value from the 10-bit space, but probably the most common one).

arichardson, 2024-10-11:
Just to clarify, it looks like this is the data for incoffset? I would imagine the values used for setbounds are somewhat different.

tariqkurd-repo, 2024-10-14:
https://github.com/riscv/riscv-cheri/pull/418 is merged, so can we close this now?

arichardson, 2024-10-14:
I agree the immediate issue is fixed, but maybe we should keep this open as a future enhancement to allow for larger values?

nwf, 2025-09-19:
In some further discussion today, I'd like to add "`YBNDSW` with `rs1=0`" as further redundant encoding space.

arichardson, 2025-09-19:
> After some out-of-band discussion today, I'd like to add "`YBNDSW` with `rs2=0`" as further redundant encoding space. (And "`YBNDSW` with `rs1=0`", too, but for different reasons than the encoding of static known lengths as discussed herein.)

That is an interesting point - we could reuse that immediate and reuqire use of YBNDSW with x0 to create zero size capabilities. I wonder if we should reserve it for a future common value (e.g. 512/1k/2k/4k) based on some further benchmarking?

nwf, 2025-09-19:
@rmn30 (with a bit of editing on my part; brilliance his, mistakes mine) suggests
```
s ? ((uimm + 3) << 4) : (uimm + 1)
```
which gives us 1 through 32 (inclusive) and then multiples of 16 from 48 (inclusive) through 544 (inclusive).  0 is available as `YBNDSW` with `rs2=0`.

arichardson, 2025-09-19:
> [@rmn30](https://github.com/rmn30) (with a bit of editing on my part; brilliance his, mistakes mine) suggests
>
> ```
> s ? ((uimm + 3) << 4) : (uimm + 1)
> ```
>
> which gives us 1 through 32 (inclusive) and [...]

I was concerned about introducing an immediate that isn't literally encoded (or a shift of that literal), but it looks like zibi is now introducing one that is `imm==0b11111?-1:imm+1`, so having addition might be a good way to improve immediate utilization.

tariqkurd-repo, 2025-09-22:
Interesting - it' s a 5-bit immediate so the adders will be very small so I think that's probably ok.
Can we show good utilisation on the bounds from SPEC which @jonwoodruff has above?
Tha table above shows the number of bits - but I don't know if the actual values match the requested ones.......
I would say that if we're going to change this then now is the time......

davidchisnall, 2025-09-22:
We've started doing some work here.  We use csetbounds to materialise globals on CHERIoT so make heavy use of this instruction and have been trying to determine if we need the 11-bit-immediate csetboudns in Xcheriot2 or if we can use this, either as is or with some small tweaks.

rmn30, 2025-09-22:
From a very small, quick study I got the best coverage just by using all integers from 0-64 (two instructions with 5-bit immediate + ybdsw with rs2=0). We need to do a bigger study to confirm and it's obviously specific to the CHERIoT ABI but suggests a quantitative approach is worth doing.

tariqkurd-repo, 2025-09-22:
0-63? that certainly would be simpler.
I think this is probably a sensitve area, as bounds setting is one of those things CHERI does which legacy RISC-V doesn't, so minimising the overhaed will help with performance results.

it covuld be that this is all noise anyway and we just go for the simplest......?

nwf, 2025-09-22:
> Weren't we already "misusing" an R-type encoding and replacing rs2 with an immediate? Is replacing both rs2 and rs1 worse?

There's precedent for stealing one elsewhere in RISC-V, with the shift instructions (`slli`, `srli`, `srai`) and (at a lightly educated glance) in the Vector goo.  There's also precedent for requiring `rs1 = rd` in the compression scheme.  But I am unaware of precedent for doing both at once or otherwise stealing both `rs1` and `rs2` in nominally R-type instructions.