riscv/riscv-cheri #410: "Zstid" CSR inconsistency

stefano-codasip, 2024-10-04:
In Chapter "Zstid" Extension for Thread Identification, under "Control and Status Registers (CSRs)" paragraph, the following statement is reported:
> Zstid adds two new CSRs to implement a trusted thread identifier (TID) used in compartmentalization. These CSRs are listed in Table 29, Table 30, Table 31 and Table 32.

So how many are the CSRs added by Zstid? Is it two (which ones) like the first statement states, or is it four as the second statement lists four of them?

tariqkurd-repo, 2024-10-04:
it's the number of implemented CSRs - I'll do a single word fix

tariqkurd-repo, 2024-10-04:
added in : https://github.com/riscv/riscv-cheri/pull/409

I'll keep this open for a while to fix other minor things

closed 2024-10-04 by tariqkurd-repo