riscv/riscv-cheri #411: zcherilevels issues

chrehrhardt, 2024-10-08:
- It seems that there is some wording missing that CL, SL and EL should be set in the infinite capability if zcherilevels is supported.
- The spec envisions LVLBITS > 1 but the current layout of the ACPERM bitmap does not envision this. Increasing LVLBITS would move existing permission bits (C, SDP) which is a no-go once software out there relies on the spec. Alternatively, SL and CL would have to be split within the ACPERM bitmap.
- Finally, if zcherilevels is not supported ACPERM will probably report zeroes in the fields for CL, SL and EL. It would feel more consistent (to me) if CL was specified as hard-wired 1 in this case. This would give a consistent set of permissions if interpreted as if zcherilevels was supported (all caps global, neither store local nor elevated load allowed, i.e. all stores keep the tags).

arichardson, 2024-10-10:
> * It seems that there is some wording missing that CL, SL and EL should be set in the infinite capability if zcherilevels is supported.

Fixed in 972e8ffa69d349d3c756e56cf581338bd940d993

> * The spec envisions LVLBITS > 1 but the current layout of the ACPERM bitmap does not envision this. Increasing LVLBITS would move existing permission bits (C, SDP) which is a no-go once software out there relies on the spec. Alternatively, SL and CL would have to be split within the ACPERM bitmap.

There is a note in the extending to more than one bit section:
> NOTE: The layout of the <<ACPERM>> input / <<GCPERM>> result is not yet defined, but existing bits will not be moved around so the <<sl_perm,SL>>/<<section_cap_level,CL>> fields will be non-contiguous.

I don't believe having them contiguous is that important since you'd probably want a "decrement" cap-level/store-level instruction once you have more than one level (or rather a `restrictperm cd, cs, <imm>` that allows things like "make this deeply read-only", "reduce the level", etc.).

> * Finally, if zcherilevels is not supported ACPERM will probably report zeroes in the fields for CL, SL and EL. It would feel more consistent (to me) if CL was specified as hard-wired 1 in this [...]

I'm not quite sure about this one - while code behaves as if all caps implicitly had a value of 1 here, reporting this could be confusing? I guess software could try running acperm and see if it actually gets cleared to determine if the levels extension is implemented?

@LawrenceEsswood @davidchisnall @nwf what do you think makes the most sense?

chrehrhardt, 2024-10-14:
Thanks for the clarification. As far as I am concerned this is resolved with the merge of https://github.com/riscv/riscv-cheri/commit/972e8ffa69d349d3c756e56cf581338bd940d993. From a software perspective it would indeed be nice if one code probe for zcherilevel by trying to clear the level but this is not a big deal.

arichardson, 2024-11-04:
I believe this issue should be addressed.

closed 2024-11-04 by arichardson