riscv/riscv-cheri #426: ACPERM RV32 order of rules inconsistency

jamie-melling, 2024-10-17:
Depending on the order in which you carry out the RV32 ACPERM rules you can get different results.  This does assume that the rules are cascaded, which is not explicitly stated.

Contrived example:

The MXLEN=32 additional rules are:

2. Clear C-permission and X-permission if R-permission is not set.
3. Clear W-permission if C-permission is set, but LM-permission is not set.

Example 1

| R | C | LM | W |  |
|--------|--------|--------|--------|--------|
| 0 | 1 | 0 | 1 | After & with rs2 |
| 0 | 0 | 0 | 1 | After RV32 Rule 2 - Clear C-perm |
| 0 | 0 | 0 | 1 | After RV32 Rule 3 - Does not clear W-perm |

Example 2

| R | C | LM | W |  |
|--------|--------|--------|--------|--------|
| 0 | 1 | 0 | 1 | After & with rs2 |
| 0 | 1 | 0 | 0 | After RV32 Rule 3 - Clear W-perm |
| 0 | 0 | 0 | 0 | After RV32 Rule 2 - Clear C-perm  |

To solve this, the spec needs to be explicit about if and when the rules are cascaded and that in such a case, the ordering of the rules should be enforced.

tariqkurd-repo, 2024-10-17:
interesting, I see no reason _not_ to cascade the rules

tomaird, 2024-10-18:
Applying them all exactly in order doesn't quite work either
Currently this rule (for all architectures):

> 3. Clear M-bit unless X-permission is set

is above this rule (for RV32 only):

> 4. Clear X-permission if X-permission and R-permission are set, and C-permission and W-permission are both clear.

But if we apply 3 before 4, then we can end up with X-permission cleared but M-bit set, which isn't valid.

I think there needs to be a deeper thought about the dependencies between these rules

closed 2024-11-04 by arichardson