riscv/riscv-cheri #428: Zcherilevels RV32 stripping EL clarification

jamie-melling, 2024-10-21:
If you have a quadrant 1 capability with perms[2:0] equal to either 0-1 or 4-5 (with w-perm) and remove just `EL`, there are two possible outcomes, neither of which break monotonicity. Either you stay in quadrant 1 and strip `C` and `LM`  to get an `R`, `W` and `X` capability.  The other option is to create a quadrant 2 capability where you strip `LM`, `W` and `X` to get an `R` and `C` capability.  The rules laid out in ACPERM do not cover which mutation is desired.

One other note, small spec bug in ACPERM rules.
[image]

tariqkurd-repo, 2024-10-21:
It looks like these rules need a thorough review.
Here: https://github.com/riscv/riscv-cheri/pull/421 we default to RC permission where there is a choice, so maybe we shoudl do the same here.

tomaird, 2024-10-22:
Typo fix here: #429

jamie-melling, 2024-10-22:
One other thing worth double-checking, in ACPERM, there is

> If the AP and M-bit field in cs1 could not have been produced by ACPERM then clear all AP permissions and the M-bit, and skip the next step

and further down on the ACPERM section

> Even though being included here CL is not considered an architectural permission.

So does the ACPERM instruction set the `CL` field to 0 when `cs1` could not have been produced by ACPERM?

It is implied that GCPERM will return 0's if the same check fails for that instruction

> If the AP field cannot be produced by ACPERM then all architectural permission bits in rd are set to 0.

closed 2024-11-04 by arichardson