riscv/riscv-cheri #453: Mixture of British and American English spellings

francislaus, 2024-11-26:
While reading through, I have come to realise that the document has both British English and American English spellings. After talking to @tariqkurd-repo, we concluded that the document should converge to American English.

marnovandermaas, 2024-11-26:
Why do we need to converge on this? Can't we let people use the spelling that they prefer? If we're going to prescribe this are we standardize the rest of the writing style of the document?

jrtc27, 2024-11-26:
Because it's unprofessional for a standards document (or any publication) to use inconsistent spelling. And yes, the writing style in such a document should be at least somewhat standardised.

tariqkurd-repo, 2024-11-26:
And the RISC-V specs use American English, so we need to be consistent

PR #454 (2024-12-12): Converging British English spellings to American English spellings
This PR is for converging British English spellings to American English spellings. This should close #453
commit b9656e13 (2024-12-12): Converging British English spellings to American English spellings (#454)

This PR is for converging British English spellings to American English
spellings. This should close #453