riscv/riscv-cheri #458: Data trace and mutating loads

tariqkurd-repo, 2024-12-03:
Using LC to access memory can cuase the value loaded into a register not to match what was in the memory, e.g. reducing CL or removing LM+W permission.

What should data trace show in these cases?

1. The actual bits from the memory
2. The value which was loaded into the register

It seems reasonable that 2. is easier to implement, otherwise you need to hang on to the original and modified versions of the data.

Any opinions?

@tomaird @jamiemelling

tomaird, 2024-12-04:
I would argue that the purpose of the data trace is to monitor what is loaded from / stored to memory. The modification of permissions is part of the instruction semantics of `lc`, which happens after the data is loaded. By tracing the modified data, the trace encoder won’t have an accurate picture of what data is in the memory - it won't know if (e.g.) the LM-perm was clear in memory already, or if it was cleared by the `lc`. And if you do a `lc` immediately followed by a `lw` to the same region of memory where the metadata is stored, then you’ll see different results on the traced data, even though the data in the memory hasn’t actually changed at all, which doesn't seem right.

tariqkurd-repo, 2024-12-12:
https://github.com/riscv/riscv-cheri/pull/463

I went for option 2