riscv/riscv-cheri #470: CHERI PTE exceptions might need their own cause code

PeterRugg, 2024-12-10:
The Zcheripte spec implies that faults caused by CHERI PTE exceptions use the existing RISC-V load and store/AMO page fault cause codes. The ISAv9 spec has separate cause codes for these: 0x1a and 0x1b.

Are we sure that we can do without separate cause codes?

chrehrhardt, 2024-12-12:
I think it would be useful to know if a fault is capability related (either through an additional exception code or a flag in say in xtval2). If you get e.g. a write fault you always have to make the page writable. However, a OS may not want to deal with capability faults if it does not have to.

arichardson, 2024-12-12:
The normal capability faults triggered when checking bounds and permissions of a register already use a separate code to ensure the kernel can handle them correctly, and I believe we also need to so the same for PTE based faults since otherwise the revocation logic ends up being mixed in with the normal page fault handling. While I think it should be possible to make it work with extra software complexity I'd prefer having separate codes. Does anyone recall how Morello handled this?

@nwf, @jrtc27 and @jonwoodruff: could you comment on this?

closed 2024-12-12 by tariqkurd-repo

nwf, 2024-12-13:
I think that agrees with my preferences: we probably could come to the conclusion in software that the problem _might_ be, say, capability load generation related by looking at the PTE, but I'd rather the hardware just told us.