riscv/riscv-cheri #472: Make Zcheripurecap depend on Zstid

francislaus, 2024-12-10:
One of the biggest advantages of CHERI is compartmentalisation. It seems sensible to make `Zcheripurecap` depend on the existence of the `Zstid` extension.

davidchisnall, 2024-12-11:
Completely agreed (with @arichardson). CHERIoT has a rich compartmentalisation model and does not use Zstid (we may replace out use of MTDC with it in a version rebased on Zcheripurecap) but they’re orthogonal features. Zcheripurecap should remain the minimum set of features for CHERI, not a wishlist of all of the things that we might want.

PeterRugg, 2024-12-11:
Note that I'm not sure Zstid without CHERI makes sense. It's only ASR that makes the register read-only, so without CHERI it's just another integer CSR that can be written by anyone in your ring.

arichardson, 2025-02-13:
Closing this as resolved.

closed 2025-02-13 by arichardson