riscv/riscv-cheri #484: Priority of CHERI page faults

Timmmm, 2024-12-13:
It didn't matter before #480 because CHERI page faults were indistinguishable from standard page faults, but now we need to specify their priorities. There are a couple of sources of page faults that you *could* do before or after the CHERI checks:

* Standard WRXU bit checks
* Page fault if A/D bits need updating but hardware updates aren't supported

From a Sail point of view I think the easiest thing is if CHERI checks are the lowest priority; i.e. the above checks are done first.

Also there's a question of what happens if hardware A/D bit updates are supported but hardware CRG/CW bit updates are not, or vice versa. It would simplify the Sail implementation (and spec!) if they always had to be the same - either all hardware updates or none at all. Would that be acceptable?

tariqkurd-repo, 2024-12-13:
>we need to specify their priorities

The global rule is that CHERI checks come first - and anything existing in RISC-V comes after. 
Is there any reason why this wouldn't work here?

(the only counter-example is a PC breakpoint - which is higher than CHERI checks on the PC - otherwise it's impossible to debug CHERI code)

>Also there's a question of what happens if hardware A/D bit updates are supported but hardware CRG/CW bit updates are not, or vice versa. It would simplify the Sail implementation (and spec!) if they always had to be the same - either all hardware updates or none at all. Would that be acceptable?

I've already tied them together in section 5.4 (@buxtonpaul asked about this before)

tariqkurd-repo, 2024-12-13:
> Ah ok, no I think that should be fine. Is that general rule written down anywhere?

I think the correct solution here is to add the PTE page faults into the exception check information on the instruction pages, and add the statement that the PTE page faults come first.

PR to follow

closed 2024-12-17 by tariqkurd-repo