riscv/riscv-cheri #494: PCC bounds checking depends on fetched instruction

tomaird, 2024-12-19:
One of the checks for PCC is that:

> All bytes of the instruction must be in bounds

This check requires to know the size of the instruction (16 bit or 32 bit), which means the instruction must have been successfully fetched.

However the exception priority list has PCC checks _above_ instruction access/page faults. So if the PCC address is at the upper end of its bounds such that a 16 bit instruction would be in bounds but a 32 bit instruction wouldn't, and during instruction address translation there is an access fault so no instruction will be fetched, it will be impossible to know if all bytes of the instruction will be in bounds. 

The spec isn't clear on how to handle this case. Should it assume a minimum sized instruction?

jrtc27, 2024-12-19:
Our spec was that you fetch in 16-bit units, doing bounds then normal translation + access fault checks for each in turn. If you had a non-C core then you could instead just fetch the whole 32-bit instruction in one go.

tariqkurd-repo, 2024-12-19:
So the PCC check here needs qualifying to say that if no valid instruction is availabe, then a minimum instruction is checked for PCC bounds.

[image]

closed 2024-12-20 by tariqkurd-repo