riscv/riscv-cheri #510: is an executable vector

LawrenceEsswood, 2025-01-30:
In several places in the document we see the phrase 

"X is an executable vector, so it need not be able to hold all possible invalid addresses."

The first instance is PCC. What precisely does "an executable vector" mean? "All possible invalid addresses" is later clarified in the document. A quick scan of the privileged spec did not show up the term.

andresag01, 2025-01-30:
I think that expression just means "its a pointer to code".

RISC-V allows optimizations such that some CSRs (e.g. MTVEC) only hold as many bits to represent all valid addresses. So this "executable vector" cropped up to indicate that the extended capability CSR (e.g. MTVECC) could benefit from this optimization and so it needs special handling as described in the Invalid Address Handling Section. Later, someone pointed out that the same optimization could be applied to capability CSRs that hold pointers to data, so we ended up with Table 51.

I agree that the term "executable vector" is not ideal -- it reminds me of something like the reset vector of addresses. It really should just be something like "capability pointing to executable memory".

LawrenceEsswood, 2025-01-30:
Possibly "X need not be able to hold all possible invalid addresses (link)" would then be sufficient for the prose? Being a capability pointing to executable memory is not sufficient condition after all (GP registers are that too, and if the optimisation applies to data pointers too then its nothing to do with being executable at all).

Maybe in 51 we could write "Code Pointer", given the second column says "Data Pointer"?

andresag01, 2025-01-30:
> Being a capability pointing to executable memory is not sufficient condition after all (GP registers are that too, and if the optimisation applies to data pointers too then its nothing to do with being executable at all).

Actually, the condition is that the register is a CSR, only ever holds pointers to memory (i.e. the extended register only holds capabilities), and the RISC-V spec allows the optimization. GP registers do not fit that description.

Both your suggestions sound good. Would you like to propose a PR?

LawrenceEsswood, 2025-01-30:
I will do so

closed 2025-02-03 by arichardson