riscv/riscv-cheri #535: Extension names need to be finalized

arichardson, 2025-02-13:
The current extension names are not valid since Zc* is currently reserved for extensions that affect compressed instructions. The names in the ISA document are placeholders for now (unless we decide that Zc* only refers to compressed instructions if it does not start with Zcheri*).

jrtc27, 2025-02-13:
Even aside from that, Zcheripte is a privileged architecture extension so should be Sfoo. Since it specifically relates to translation, it should even be Svfoo. On top of that, Svcheripte (or Svcheri once you remove the wholly-redundant "pte" suffix) is too generic a name for adding (U)CRG. Something like "Svucrg" is probably the right name for it, which is completely different to what it's currently called.

arichardson, 2025-02-13:
> Even aside from that, Zcheripte is a privileged architecture extension so should be Sfoo. Since it specifically relates to translation, it should even be Svfoo. On top of that, Svcheripte (or [...]

Agreed, I think we should proactively use Svucrg which is clearly better than the current placeholder. Will open a PR shortly.