riscv/riscv-cheri #536: Finalize sub-exception code reporting mechanism

arichardson, 2025-02-14:
This is a question for ARC review.

In the current CHERI specification, we report an exception sub-code in mtval2/stval2/etc. The upside of this approach is that it does not impact other extensions, but the downside is that we add new CSRs that were not previously required just to report a few bits of extra information.

Would it be cleaner to reserve the high 8 or 16 bits of the *cause CSR to report this additional sub-cause? mcause >= 64 is reserved, so this could be possible? Thena CHERI-aware operating system could to check `cause & 0xffffff` in the trap handler and obtain the remaining information form `cause & 0xff000000` instead of having to read the new *tval2 register.

This would also make it easier to reuse the software check exception code instead of the custom 0x1c (#60).

Original discussion: https://github.com/riscv/riscv-cheri/pull/379#pullrequestreview-2329090208?

davidchisnall, 2025-02-14:
In CHERIoT, we have a pending patch to the RTOS that does a mapping very like this for reporting to user code. Using the low bits would be a bit cleaner since it then makes the CHERI exceptions a range. Where you can check if it’s in the range with a single and and can then extract the CHERI cause with another one.

arichardson, 2025-02-21:
I like the idea of using the low bits for the extra information. Right now mtval values > 64 are marked as reserved and we currently require 8 bits of mtval2 state (although only 4 bits are allocated so far).

If we want to keep reserving 8 bits of exception space we could say 512->1023 is reserved for CHERI, i.e. CHERI exception was triggered if `mcause & 0x200 == 0x200`.

arichardson, 2025-02-26:
I just had a look at [the CLIC spec](https://github.com/riscv/riscv-fast-interrupt/blob/master/src/clic.adoc) and it appears it already adds sub-fields to mcause, so doing this for CHERI would also be consistent.

```
 mcause
 Bits   Field         Description
 XLEN-1 Interrupt     Set to 1 for interrupts and 0 for exceptions
 30     (reserved)    Contains minhv (AKA xinhv) bit when smclicshv extension present
 29:28  mpp[1:0]      Previous privilege mode, usually same as mstatus.mpp
 27     mpie          Previous interrupt enable, usually same as mstatus.mpie
 26:24  (reserved)
 23:16  mpil[7:0]     Previous interrupt level
 15:12  (reserved)
 11:0   exccode[11:0] Exception/interrupt code
```

The current privilege spec states that anything from 64 is reserved, i.e. only bits 5:0 are allocated. Seeing as there are already new fields allocated above the exception code, maybe the cleanest approach would be to allocate the free bits 15:12 as the sub-code that reports tag violation etc.? Not sure if this is cleaner than allocating lots of values in the excode region using the[ approach in the previous comment](https://github.com/riscv/riscv-cheri/issues/536#issuecomment-2675634062)

nwf, 2025-02-26:
AFAICT, the CLIC `mcause` values will always be interrupts and so have the `Interrupt` bit (at index `XLEN-1`) set, and CHERI traps will always be synchronous exceptions and so have that bit clear, and so there is no need to contort ourselves around its choice of bitfield layout. 
 Besides, even slimmed down to what's defined and not reserved for future expansion, the existing `mtval2` bitfields amount to 2 (type) + 3 (cause) = 5 bits, which wouldn't fit in `15:12`.  I think I'd vote for [the previous comment](https://github.com/riscv/riscv-cheri/issues/536#issuecomment-2675634062)'s suggestion.

nwf, 2025-04-24:
By `software error` do you mean `software check` (18)?

I think we would prefer not to need `mtval2` in CHERIoT, because we certainly don't implement the hypervisor extension, so I still think it'd be nicer for us if we could pack the CHERI exceptions into `mcause`.

arichardson, 2025-04-25:
That sounds like a reason for not going with the software check. I'd also prefer packing the few mtval2 bits that we need into mcause. Should I kick off a discussion on the RISC-V privileged mailing list?

tariqkurd-repo, 2025-06-23:
also see: https://github.com/riscv/riscv-cheri/issues/587
mtval2 causes problems with double traps and resumably NMIs

tariqkurd-repo, 2025-07-02:
We have some ARC feedback:

```
Upon further discussion, ARC is open to using a separate exception
code or codes for CHERI faults, especially if that would avoid adding
a separate CSR.

Also, for the double trap or RNMI case, are these values required for
recovery or just to speed recovery?  In other cases, these exception
reporting registers have values that are provided to speed execption
handling, but can otherwise by more slowly recovered from the
execution context if there is no performance issue, as in recovering
from rare errors.
```

We spent a lot of time in the past debating making mtval into mtvalc, and reporting the full cap.
This causes problems because we don't want to have to do a representability check to produce the final value written to mtvalc.
E.g. a load offset makes the resulting cap unrepresentable, we can't just add the offset and report mtvalc without clearing the tag.

I'm now wondering if what we should do is this:

1. have a separate mcause for TAG and OTHERs
2. report the detagged mtvalc value - so no rep check required
3. let software work out the fault type, it will know if the tag was set or not due to the mcause value

That way we only need 2 mcause values which should be more popular than what we have now.
The other fault types (bounds, permissions etc.) woudl be decoded `slowly` by software

commit 8046b598 (2025-07-11): Rework exception reporting (#654)

Update with the ARC feedback allocating mcause values 32 onwards for CHERI faults.
Also merge invalid address and bounds fault and
stopp overloading CHERI PTE faults into normal RISC-V page faults.
We also reduce the number of faults to the minimum that needs to be differentiated: instruction, load, store plus the load and store PTE faults.
In theory we could merge the load and store access faults since any emulation will need to decode the instruction anyway, but keeping them separate is consistent with the base ISA.
This allows us to remove Xtval2.

Fixes:  [... cut]
closed 2025-07-11 by arichardson