riscv/riscv-cheri #548: Clarify what happens if "Reserved 1" bits are clear in the acperm mask

chrehrhardt, 2025-02-27:
The latest PR that introduced bits that are "Reserved 1" int the `acperm`/`gcperm` masks does not really explain what happends if one of these reserved bits is zero in the mask passed to `acperm`. At least two options are possible that I can think of:

- Make the bits WARL as most riscv registers do, i.e. trying to clear the bits is simply ignored on hardware that has them set to "Reserved 1".
- Clear the tag on the capability.

I'm not sure which version I'd prefer, though.

tariqkurd-repo, 2025-02-28:
I vote for making the reserved bits WARL, so it doesn't matter what the written value is

martin-kaiser, 2025-02-28:
If the software understands more permissions than the CPU, the software may set one of them to 0 where the default is 1.
The CPU sees an acperm instruction with a reserved1 bit set to 0. The bit is ignored anyway and does not change acperm's result.

I think this case should be supported, the reserved bits should be WARL.

jrtc27, 2025-02-28:
If software is asking to clear a bit that doesn’t exist then it is broken. Turning it into a nop could be dangerous (eg if software expects mutable load to exist and wants to clear it when delegating a capability to some other compartment, it should not silently degrade into handing out a capability that permits stores). It should tag clear because you’ve asked for something that is not supported.

chrehrhardt, 2025-02-28:
I actually tried to prepare software (linux kernel and musl) for this change under the assumption that the tag is cleared and it is really horrible. You would have to probe for supported hardware features not only in the kernel but also in musl on every program startup. In fact each and every software component, that is doing something non-trivial with permission bits that may or may not be supported by hardware would have to probe. Otherwise, clearing a permission bits you _do_ know about but that happens to be unsupported on the current hardware will invalidate your capability.

And the whole _point_ of this entire change is to ensure that the same binary can run on hardware with different features.

Thus, even though I started out being in @jrtc27 's camp I don't think it will work and I tend to agree with @arichardson that software that depends on an optional feature should check for it during runtime and refuse to even start if the feature is missing.

So if we make the bits WARL it encourages software to adopt new security features because you can just pretend that the feature is there and your software will benefit from the feature if it is supported and continue to run if it is not.
On the other hand if the tag is cleared there is no easy path to adopt a new security feature because you would either do probing and replace constant bitmask with variables or you would have to ship different binaries on hardware that has the new feature in order to make use of it, i.e. break backward compatibility.

jrtc27, 2025-02-28:
I would like to see evidence of that; I do not see how it would be as you describe.

LawrenceEsswood, 2025-02-28:
I think if software understands more permissions than hardware, there are really two cases:

1) Software needed to clear that bit to be correct and will not function correctly if it cannot.
2) Software was clearing that bit as a "best effort", if it was possible to clear.

1) would prefer tag clear, but could also check whether that bit was clearable at all as part of instantiation of the library and refuse to load if it wasn't.
2) _really_ wants to not tag clear. Otherwise you end up writing "if feature exists, clear, else don't attempt".

Because 1) can be fixed at load time, I think we should lean to not tag clearing.

chrehrhardt, 2025-03-01:
Trying to illustrate my point with an example: Assume that we forgot to specify the "load mutable" bit and want to introduce it as part of an extension. I think it is a very good example because it subdivides read permission and thus would be allocated into one of the "Reserved 1" bits. You can make similar points with levels but the levels feature is more complicated which makes it harder to get the point across.

Now, software that does not know about load mutable will assume that the bit is "Reserved 1" and will continue to work whether the hardware supports the feature or not. So just ignoring the new feature is a possible sweet spot. But it is not where we want the sweet spot to be for software vendors.

Once you want to start supporting the feature how would you clear the load mutable bit in a backward compatible way, i.e. in a way that allows you to run the same software stack on hardware that supports "load mutable" and on hardware that doesn't?

As @LawrenceEsswood points out you would have to do different things depending on hardware features because trying to clear "load mutable" on a hardware that does not support the feature will render your capability invalid. So software will have to probe hardware features in order to determine if it can safely clear the "load mutable" bit.

It gets even worse if you want the compiler to generate the `acperm` that clears load mutable e.g. based on variable annotations because the compile has no easy means to probe for the feature on startup. Instead people will end up doing an `acperm` followed by something like `cbld` to restore the tag from the original capability.

arichardson, 2025-03-13:
Would also be good to get some input from CHERIoT here. @DavidChisnall @nwf what is your position on reserved-one bits being ignored when set to 0 in ACPERM vs tag-clearing on such a request?

chrehrhardt, 2025-03-17:
If you have this software that can only really run on hardware that has optional extensions wouldn't you have to probe for the presence of these features anyway? Relying on tag clear will likely be insufficient in this case. Apart from that the result will be that your software starts and then fails in unpredictable and unexplained ways with CHERI faults at runtime. I fail to see how this is what you want.

And as you admit doing the tag clear basically ignores that someone might want to build software that runs with and without a particular feature/extension. Many of the extensions _will_ provide additional mitigations if used and you still get the CHERI bounds checks and other benefits of if you don't use a missing feature.

nwf, 2025-03-18:
I've yet to see an example of software that can run with(out) a particular permission and retain its intended security properties.  If you'll forgive a bit of an acknowledged hard-line stance, CHERI is supposed to provide _deterministic_, _well-founded_ memory safety; it is... distressing, perhaps, to contemplate proposals for running software built with a given set of assumptions in mind without said foundations in place.

More technically, if "set of expressible permissions" is part of the ABI, as I imagine it must be, then I'd expect program linkers and loaders to balk if given inputs not known to be compatible with each other or the running system, as if one tried to load a 686-requiring program on a 486, to pick a real-world example from my past.

chrehrhardt, 2025-03-19:
Well, all the existing C-software out there is such a software. It's been designed without CHERI in mind and we promise to make it (more) secure if it is run on CHERI. This improves security simply by having the bounds checks it doesn't even _need_ any permission bits to prevent a substantial class of exploits. I'm not arguing that we shouldn't have permissions. But I do claim that even if hardware didn't have any permissions the bounds checks alone would be beneficial for software security. Compared to the non-CHERI version of the software that would still _increase_ security and not reduce it.
IMHO if you are relying on CHERI for the security of your software you are doing something wrong! Software should never even attempt to perform an operation that CHERI prevents. If this happens your software is buggy in the first place.
Thus the provable security properties your are talking about are only really relevant when a malicious actor has already gained control or is about to gain control it should not have. In this case CHERi can either prevent the exploit or if the exploit already happened it can limit its scope.
If some of the permissions are not available you may not be able to _proof_ certain properties of these limits but the permissions that are present will still prevent some types of attacks.
So if you'll forgive a bit of a commercial argument: If we want CHERI to succeed we have to make it as easy as possible for people to adopt it and saying I don't care about downward compatibility is quite the opposite. It is already very unfortunate that we cannot build CHERI enabled software that runs on non-CHERI systems. However, we should at least make the adoption of new security features possible and as easy as possible.

Having said that, this discussion clearly shows that the spec need clarification one way or the other.

jrtc27, 2025-03-19:
> IMHO if you are relying on CHERI for the security of your software you are doing something wrong!

But that's the whole premise of CHERI compartmentalisation, and it's things like "hand out a restricted pointer to some other compartment" where the question of "what should happen if you try to clear a permission that doesn't exist?" starts to become a very sensitive question, and why Wes and I are pushing back on the idea of turning invalid requests into no-ops.

Also, what happens if you start running on a system that uses a permission bit for something *other* than the extension you think *might* exist? Then your software blows up (hopefully). You cannot avoid the fact that you need to check whether the thing you want to use actually exists, and if it doesn't you have no idea what it may do, so you must not do it, and thus it is completely reasonable, and the safest option, to make the hardware tag clear if you ask for something that isn't implemented.

chrehrhardt, 2025-03-20:
So taking a step back: With the introduction of the "Reserved 1" bits we've suddenly introduced a surprising non-intuitive effect of `ACPERM` and `GCPERM`: It is possible that `GCPERM` returns permissions that cannot be cleared. Maybe this is actually not what we want. Basically what we want current software to do (if it cares about forward compatibility) is _preserve_ certain permission bits unless it knows what they mean. Preserve generally implies a read-modify-write cycle but with `ACPERM` this is implicit because we can never set bits and pass in a mask of bits to keep.
So one solution might be to define the bits as "Reserved Preserve" instead of "Reserved 1". This would imply that the bits are _clear_on hardware that does not support them in the bitmask reported by `GCPERM`. However, software is required to set the bits to "1" in any mask used for `ACPERM`.
Sticking with the "load mutable" example from above this would have the slightly confusing effect that the combination "read is set" and "load mutable is clear" in `GCPERM` would mean different things on hardware that has load mutable. However, I don't think that is a problem.
If software is written without knowing about load mutable and runs on hardware that does have it it will never look a the load mutable bit and assume that it has load mutable permission anyway.
On the other hand software that is written with load mutable in mind running on hardware that doesn't have it is the case you don't care about. If you do anyway you will have to look at the infinite cap you are given and refuse to run if it does not have the load mutable bit set. Or you continue to run and accept the fact that all loads are mutable.
Is this a viable solution?

jrtc27, 2025-03-20:
It’s reserved 1 so CBuildCap works.

nwf, 2025-03-20:
To expand that a bit (@jrtc27 please complain if I've misunderstood): "It's reserved 1 so CBuildCap works in upwards-compatibility deployment scenarios, where hardware knows about dependent permissions that software does not".

chrehrhardt, 2025-03-20:
But wouldn't that require "Reserved 1" bits in the in memory capability representation? We only have the reserved bits in the `acperm`/`gcperm` bitmaps. But I'm sure I'm missing something obvious here, again?

tariqkurd-repo, 2025-09-05:
given that we've changed the spec of ACPERM to be CLRPERM (YPERMC) is this still an issue?

chrehrhardt, 2025-09-05:
No, this is no longer an issue with YPERMC.

jrtc27, 2026-09-21:
This is in fact still an issue. The switch from ACPERM to CLRPERM (YPERMC) doesn't change the problem, all it does is invert the input. The current spec does not say what happens if you try to YPERMC with a mask that includes one of the Reserved 1 bits, and IMO should treat that as a request for an illegal permission combination (presumably tag clear and either preserve the rest of the bits or force them to 0 like integrity checks).

tariqkurd-repo, 2026-09-22:
It gets close by having a NOTE saying "Future extensions may include hardwired permission bits, in which case they are not cleared by set bits in rs2" but doesn't say anything about _existing_ hardwired permissions which is an oversight, and so don't appear to be handled as an illegal combination of permissions.

So what do we do:

1. Change the text to say "Hardwired permission bits are not cleared by set bits in rs2" and get rid of the NOTE, so that those bits of the mask are ignored?
2. Invalidate the output as Jessica suggests if the mask isn't 1 for every bit hardwired to one?

@chrehrhardt any opinion? Both are very simple but would there be implications from either?

chrehrhardt, 2026-09-22:
Well, the situation hasn't changed very much:

I think you should be allowed to "speculatively" clear reserved bits because that is the only way to make use of the new feature on hardware that supports it while maintaining binary capability with hardware that doesn't. And that again facilitates adoption of the new security feature. If your program is incorrect in the absence of the new feature you will have to do some kind of probing at startup. But I suppose you have to do that anyway because that bit that you can clear could be something completely different.

Obviously, @jrtc27 thinks otherwise.

On a slightly related note:
The example in the spec to justify "reserved 1" bits (finer-grained ASR permissions) is not really convincing to me. If we implement finer-grained ASR permissions you most likely want to _disallow_ access to all but a very specific set of MSRs. E.g. only allow access to the timer MSRs but to nothing else.

jrtc27, 2026-09-22:
> the only way to make use of the new feature on hardware that supports it while maintaining binary capability with hardware that doesn't

`if (<code to probe for RISC-V extension>) ... else ...`? One should have to be explicit in wanting the "use it if it exists" behaviour, not have to be explicit in wanting the "scream at me if I try and use something that doesn't exist" behaviour.

jrtc27, 2026-09-22:
One can have opinions on which way round is better, but please do not go asserting it only *works* your way round when that's quite clearly not true.

chrehrhardt, 2026-09-22:
I was trying to be at least somewhat concise not dogmatic. Sorry. I do believe that with tag clearing software where binary compatibility matters will tend to ignore the new feature.

jrtc27, 2026-09-22:
But if you still have to consider the fact that another extension might assign a different meaning to that bit that you're speculatively clearing, do you not already need exactly the same code as you would need if attempting to clear reserved 1 bits cleared the tag?

nwf, 2026-09-22:
FWIW, I still think I was right above:

> All that said, I guess the position I'd take on this particular point is that an ACPERM with a 0 bit in a reserved-1 position ought to clear the tag in the result: having been asked to shed a permission we can't articulate, the maximal expressible subset of the intersection is no permissions. (Dually, ACPERM with a 1 bit in a reseved-0 position does not necessitate tag-clearing, as we're being asked to preserve a permission we can't articulate, and so is by definition absent in the precursor's set.)

chrehrhardt, 2026-09-22:
In either case the chances that this happens are tiny. But the risk assessment is different IMO:
In the case where you are opportunistically trying to use a security feature and that feature's  bit was allocated to something else you either won't notice because the unknown permission is irrelevant to you or you will blow up because you run on hardware you are really not made for. That's fine in either case because chances that this happens are small.
On the other hand if you have a program that absolutely doesn't want to run if the feature is missing, you might not want to ignore the possibility that the bit can be cleared but in fact does something else than what you expect.

nwf, 2026-09-23:
More generally: _please_ don't consider the behavior of _particular, bit-starved instructions_ to be a viable mechanism for probing for architectural behavior.  I've said this before and will continue to do so: programs get loaded and run **in a context** by other code (or, in the base case of firmware, by a human).  It is the responsibility of that contextually-aware loader to ensure that the loaded code can run.  There are ways of defining on-line interrogation of an ISA and its extensions; "just run some code and see what happens" is not a viable approach.

arichardson, 2026-09-23:
**TL;DR:** IMO both behaviours have valid use cases, similar to `YBNDSRW` (best-effort) vs `YBNDSW` (must be exact) for bounds. However, we are in a frozen state and unless this is a critical bug (which I don't believe it is), we shouldn't change the Sail semantics now. Right now the behaviour is only in the sail and a non-normative NOTE. We should make the note normative. If there is demand for strict checking in hardware, add a `YPERMC_EXACT` instruction in a future extension (and until then it can be emulated with a YPERMC/YPERMR sequence).

----

I was convinced we already specified this but it turns out the latest spec is a bit vauge. So I did a bit of git digging: Privously, the text actually said *"Some permission bits may be hardwired, in which case they are not included in this bitwise AND"*. That really should be normative text rather than a `NOTE:` from the start, and it was then narrowed to *"Future extensions may include..."* in #611 when `Zcheri0lvl` was removed.

Personally, I have a minor preference for the current ignore semantics over tag clearing but 
I don't feel strongly about which approach is better.

The main software case I'm worried about with tag clearing is something like `cheri_perms_and(c, READ|CAP)` to hand out a read-only capability. In the current semantics clearing everything except `KEEP_PERMS` requires setting mask bits to 1 for all permissions to be removed. With tag clearing, you can't just pass `~(KEEP_PERMS)` because that sets reserved bits to 1 and immediately tag-clears on today's hardware.

To make that portable to a !Zylg system, software would have to avoid masking the base `RES1` bits (`~(READ | CAP | RES1_BASE)`). But on such a system, the `Zylg` bits (`LG`, `SL`, `GL`) are part of `RES1_BASE`. That means when that portable binary runs on a `Zylg` system, it won't clear `SL` or `GL`, so you don't actually get "only `READ|CAP`". The same problem applies to any future extension permission.

In theory the compiler or `cheriintrin.h` could implicitly OR in the `RES1` bits based on `-march`, but that seems like a bad idea: it silently alters the mask behind the programmer's back, doesn't help assembly, and a portable binary built for `-march=rv64y` still won't clear `LG`/`SL`/`GL` on a Zylg system.

Also I don't believe the current semantics (which I believe do have their uses) can be emulated in software:
- With ignoring of bad bits, software that needs strict checking can check once at startup, or per-call with `(cheri_perms_get(cd) & clear_mask) != 0` and clear the tag / trap. If we really wanted to, the compiler could default to emitting the tag-clearing sequence for the intrinsic if we wanted strict checking by default in C.  A future extension could optimize that with a conditional tag-clear instruction or `YPERMC_EXACT`.
- With tag clearing, relaxed checking cannot be emulated without dynamically querying implemented permissions at runtime and masking `rs2` before `ypermc`, which makes immediate masks impossible for portable code.

Strict checking can be emulated on top of ignoring bad masks, but relaxed checking cannot be emulated on top of tag clearing. It also gives the property that on an unsealed, valid capability, `YPERMC` never clears the tag regardless of the value in `rs2`.

tariqkurd-repo, 2026-09-23:
> To make that portable to a !Zylg system, software would have to avoid masking the base RES1 bits (~(READ | CAP | RES1_BASE)). But on such a system, the Zylg bits (LG, SL, GL) are part of RES1_BASE. [...]

So you'd have to specifically know which permissions are reserved one for the given extensions to make this work, instead of just stating which permissions youw ant to _keep_ and not worrying about reserved ones.

>Strict checking can be emulated on top of ignoring bad masks, but relaxed checking cannot be emulated on top of tag clearing.

Certainly true - the strict checking can always be implemented in software.

This sounds sensible to me.

For the portability issue, software can always test for reserved one bits by using YPERMR on the NULL_CAP and then OR those into the mask, but it would be better not to have to do that and just ask for the permissions you want to keep.

tariqkurd-repo, 2026-09-23:
The portability argument is the most compelling one, and it's also true that more exact checking can be done in software if the reserved bits are ignored in the mask, so I'm in agreement with @arichardson - it looks like the correct solution.

jrtc27, 2026-09-23:
I have a lot of thoughts of disagreement to write up, but today was busy with meetings and I'm not working the rest of the week. I think it would be good to have a proper call about this to discuss before any decisions get made either way (and, to be clear, my perspective is that any decision is a spec change, because the spec does not say what to do, and just because the Sail model does one thing doesn't mean that's the only allowed behaviour, or even correct behaviour, since we're not yet at the point where the model is authoritative).

arichardson, 2026-09-23:
> I have a lot of thoughts of disagreement to write up, but today was busy with meetings and I'm not working the rest of the week. I think it would be good to have a proper call about this to discuss before any decisions get made either way 

I agree that discussing this in a call rather than using github issues on how to proceed here is probably a good idea. And the outcome of that discussion could be that we need the spec change, but we need to be extremely certain that this is an absolutely critical bugfix since we are in the final stages of review.

> (and, to be clear, my perspective is that any decision is a spec change, because the spec does not say what to do, and just because the Sail model does one thing doesn't mean that's the only allowed [...]

I would argue the current spec does define the "best effort non-tag-clearing" behaviour. I do agree with you that the way it's written now **is extremely poorly worded** and needs to be clarified, and we can discuss the right behaviour on the call.

The reason I say it mandates the non-clearing behaviour is that the steps say the only way to get a tag=0 result back is if the input is sealed or corrupted.
So the way I read the current spec, if you pass an `rs2` mask with a `Reserved 1` bit set to 1:

- Step 1 unpacks `{cs1}`'s `AP` and `SDP` into the intermediate bit field, and Step 2 clears the bits where `rs2` is 1 (so the `Reserved 1` bit becomes 0 in the intermediate bit field).
- Steps 3-4 then take the architectural permissions and `SDP` from that intermediate bit field, apply the transition rules, and encode the resulting `AP` and `SDP` fields into `{cd}`. I would argue that the `Reserved 1` bit positions don't map to any bit of `AP` or `SDP`, so clearing them in the intermediate bit field has no effect on what gets written to `{cd}` (so `YPERMR` reports them as hardwired 1 again).
- Steps 5-6 only ever set `{cd}.tag=0` if `{cs1}` is sealed or fails integrity checks (and even when a requested permission combination isn't supported by the encoding, Step 3a returns *"a minimal set of permissions, which may be no permissions"* rather than clearing the tag).

Promoting the NOTE to real normative text and fixing the wording of the remaining steps would make this much clearer, but tag-clearing on an unsealed valid capability would definitely be adding a new rule.

tariqkurd-repo, 2026-09-24:
we will make a spec release early next week to continue teh ARC review, so any discussion will ideally be on monday

arichardson, 2026-09-29:
Sorry for closing this one, I removed the Fixes: trailer from that PR but somehow github still closed it.