riscv/riscv-cheri #558: What happens if PTE.CW or PTE.UCRG are set and CRE=0?

tariqkurd-repo, 2025-03-13:
If CRE is 0, so software is running in effecitvely "RISC-V mode" with PCC/DDC checking, then all CHERI features are _reserved_ i.e. encodings, CSRs liek DDC etc.

We haven't (clearly) specified what happens if software sees non-zero PTE.CW/PTE.UCRG bits in these cases.

I'm guessing that they shoudl be handled as _reserved_ and so there'll be a page fault / access fault (whatever is normal for a bad PTE entry) but it's worth a discussion about this.

jrtc27, 2025-03-13:
CHERI *features* are not reserved. CHERI *state* is. The premise of this is false and making this change is inconsistent with the rest of RISC-V.

jrtc27, 2025-03-13:
> If we run a hypothetical "RVA2x" compliance test suite that checks that all reserved instructions trap and reserved bits in the PTE fault

No? To quote profiles.adoc:

> Extensions that are not explicitly listed in the mandatory or optional categories are termed non-profile extensions, and are not considered parts of the profile. Some non-profile extensions can be added to an implementation without conflicting with the mandatory or optional components of a profile. In this case, the implementation is still compatible with the profile even though additional non-profile extensions are present. Other non-profile extensions added to an implementation might alter or conflict with the behavior of the mandatory or optional extensions in a profile, in which case the implementation would not be compatible with the profile.
>
> NOTE: Extensions that are released after a given profile is released are by definition non-profile extensions. For example, mandatory or optional profile extensions for a new profile might be prototyped as non-profile extensions on an earlier profile.

It is completely valid to implement additional extensions. CHERI would be a set of non-profile extensions, and would be non-conflicting (so long as you have and default to integer mode). You do not need to trap on every aspect that isn't part of the profile itself.

arichardson, 2025-03-13:
Ah thanks, was not aware of that. In that case I'd say we can just keep these bits as valid unconditionally since in CRE=0 operation the value of them does not matter.

@tariqkurd-repo

arichardson, 2025-03-17:
Fixed in https://github.com/riscv/riscv-cheri/commit/cf51732e0bf6b2775516d168642f05dcca54652d

closed 2025-03-17 by arichardson