riscv/riscv-cheri #565: RV32 is it impossible to clear only write permission?

veselypeta, 2025-03-17:
Looking through the rules of `acperm` it seems to me that it might be impossible to only clear write permissions.

Consider having a rv32 capability with full permissions and with `zcherilevels` disabled. If you use `acperm` with a bitmask to clear only the write flag, then the following rules will apply:
1. (Rule 1) - Clears `ASR` - this is fine
2. (Rule 10) - Clears `LM` - as `W` is cleared and `EL` is not available.
3. (Rule 13) - Clears `X` - as `LM` is not available but `C` is still set.

This is a shame as clearing `W` permissions from a capability is quite common for initializing function pointers, which will need `X` permissions. Also there is an ideal encoding for these permissions in the table (Q1 entry 2/3), but it seems the rules for `acperm` might prohibit getting to this for rv32.

arichardson, 2025-03-17:
Hmm, IMO removing W should transition to from infinite to "Execute + Data & Cap RO" - if this is not what happens following the rules we should definitely fix that.

nwf, 2025-03-17:
FWIW, that's what happens in CHERIoT's permission structure.  We don't have an infinite cap, but starting from $\top_M$ (the top of the "mem-cap-rw" substructure, bearing CHERIoT/Zcheri permissions: LD/R, LG, LM, MC/C, SD/W, SL) and clearing SD/W, the operation also clears the dependent SL, and the result is the top of the "mem-cap-ro" space (bearing LD/R, LG, LM, MC/C).  (See page 64 of https://cheriot.org/cheriot-sail/cheriot-architecture.pdf for details.)

LawrenceEsswood, 2025-03-17:
The problem here is assuming in the absence of zcherilevels that the EL permission is effectively 0 for the purpose of applying the rules.

Clearly it is not helpful having the rules reference a permission that does not exist, but the actual intended behaviour is that the rules apply as if it did (so, in this case, inf cap has EL=1 and so 10 keeps LM).

I think we have two options to make this clearer.

Option (A) Rephrase the rules (possibly a third column) to not mention EL at all if we do not have `zcherilevels`:
Remove 7
Remove 8
Replace EL in 10 and 13 with (C & R & (LM | RV64)) (note: this is just in-lining rules 7 and 8)

Option (B) Admit that it exists for the purpose of the rules (and can be modified for the purpose of interactions between the rules), but is hardwired 1 at the start and end of the rules. Therefore the current note would be correct in that EL cannot be modified (it always starts and ends as 1 when applying ACPERM), but the intended interaction between 7/8 and 10/13 makes sense.

arichardson, 2025-03-17:
> The problem here is assuming in the absence of zcherilevels that the EL permission is effectively 0 for the purpose of applying the rules.
>
> Clearly it is not helpful having the rules reference a [...]

I think Option B) is simpler conceptually: When Zcherilevels is not implemented, we evaluate these rules in order as if EL/SL were set at the beginning. If the rules end up zeroing EL/SL, we continue evaluating with that value, but at then end of the evaluation EL/SL are set back to the reserved 0/1 value.

jrtc27, 2025-03-17:
CHERI without Zcherilevels should just not talk of levels at all. It's up to any extension on top of base CHERI to say how it interacts with everything. Zcherilevels just happens to be an extension defined at the same time as base CHERI.

jrtc27, 2025-03-17:
If it's written that way for simplicity that's fine. But logically that's how it should behave, and if you find that the spec needs to be contorted in weird ways to smash it all into one definition then that's a sign it's the wrong approach, or that the smashing together needs to be reworked, such as qualifying with "if X is implemented then".

tariqkurd-repo, 2025-03-18:
ok - so we need to do two things:

1. the rules should _assume_ that EL/SL are set at the start if Zcherilevels is not implemented
2. we need to structure the documentation sensibly which doens't assume that Zcherilevels exists (ideally)

the only question now (if people agree so far) is _how_ to update the spec

closed 2025-03-26 by tariqkurd-repo