riscv/riscv-cheri #568: consider including Zcherilevels in the base ISA

tariqkurd-repo, 2025-03-24:
We'll have an easier time with the CHERI ratification if we include Zcherilevels in the base ISA.
In general reducing options is good.

Is there any reason not to just include it?

Do we think the spec is mature enough? The hardware cost is very low.

Is ti similar enough to the version in CHERIoT for us to have confidence that it's correct, and also we have several months until spec freeze - so we still have time to find and address any issues.

nwf, 2025-03-24:
FWIW, the comparison to CHERIoT and research-CHERI only for the two-level version.  In particular, AFAIK, there has been no work demonstrating the practical sensibility/utility/viability of multi-level schemes.

From hallway chatter that I recall, there's at least suspicion that a _linear_ ordering of levels is perhaps not the right one, or not the only one, that we will want; it seems at least moderately likely that software will want orthogonal L/G-like IFC (candidates include kernel-/user-space caps and a distinction between (un)managed heap memories).

jrtc27, 2025-03-24:
It is not clear that every implementation will want a level scheme that looks like this (or a clean extension thereof). Making it optional allows its omission or replacement in such environments.

tariqkurd-repo, 2025-03-25:
This is true - but it means we may have to ratify twice - once without it and then again for the extension.
We need to get a 100% perfect description of the ISA into the RISC-V ISA manual.
If we omit Zcherilevels, then we'll have to do that work again with it included, which will be effort, and requires review etc.

We _can_ omit it, but it will be a lot more work to add it later than just to include it.

There'll definitely be two versions of the ACPERM instruction, for example, as Zcherilevels redefines the rules for removing permissions.

tariqkurd-repo, 2025-04-23:
This came up inthe review with Krste. He pointed out that the hardware cost is very low, but the complexity of making it optional in the specification is very high so we should just include it.

He was quite clear - choose something to ratify and ratify all of it with minimal options.

Therefore I propose mandating the 2-level scheme and leave more levels for future expansion.

davidchisnall, 2025-04-23:
I am quite nervous about building something into the base specification with zero practical experience of how software would use it.  Zcherilevels is a research project.

tariqkurd-repo, 2025-04-23:
Fair comment, and the Zstid extension is similar.
I think so either we delay freeze until we're confident of the spec, or we drop them from the spec and pursue them later, so really this comes down to how we handle the ratification process.

Krste was very clear - choose a configuration and ratify that - so we either take more time and include more features or we go quickly and then have to repeat the process once we add more features.

nwf, 2025-04-24:
FWIW, Zstid is significantly less "research".  At the very least, it doesn't have configuration knobs like `Zcherilevels` and we know, concretely, how we might use it if we had it in CHERIoT (as the active compartment's global data pointer; see #393).

arichardson, 2025-04-24:
> I am quite nervous about building something into the base specification with zero practical experience of how software would use it. Zcherilevels is a research project.

Would you want to ratify 1-bit level as part of ChERIoT instead of making it mandatory for RV32Y? I would really like to have 1 bit since that has been shown to be useful.

davidchisnall, 2025-04-24:
I agree.  I'm less convinced that we want `Zstid` in CHERIoT, but the idea of having a thread pointer register is a pretty well-understood thing and the change to make that a capability is quite small.  I see three categories of thing:

 - Things that every CHERI system will definitely want (these should be in the base spec).
 - Things that some CHERI systems will definitely want (these should be optional extensions that are ratified at the same time or shortly after the base spec and are part of one or more profile).
 - Things that CHERI systems *might* want.

Core things like loads and stores, permissions and so on, are in the first category.

Zstid is in the second category.  We probably don't want it *everywhere*.  A CHERI GPU will want something quite different for tracking threads because it will have a very different model.  CHERIoT probably doesn't want it, but might want it plus some additional loads and stores that use it as a base (we currently remove almost all `AUICGP` instructions with a linker relaxation, so moving CGP to use a CSR instead of a GPR is less than ideal but we might be able to build something on top of Ztid that would be more efficient than what we do today).

Zcherilevels is the third category.  We don't have any software using it and the software that uses the local/global stuff the most (CHERIoT) doesn't want quite what's in Zcherilevels (though I think we can squeeze it enough to make it look like Zcherilevels if we need to, to reduce our diffs to other extensions).

davidchisnall, 2025-04-24:
> Would you want to ratify 1-bit level as part of ChERIoT instead of making it mandatory for RV32Y? I would really like to have 1 bit since that has been shown to be useful.

I think, with the most recent set of changes, the 1-bit version can express the same things, but the way that it's expressed architecturally is less clear and we'll paper over the differences in software to make it not really look like levels (especially with respect to sealing).  

It's odd to me that we would ratify levels and not sealing, given that we've got quite a lot of software that uses sealing (the compartmentalised rtld uses it, CHERIoT uses it, the CHERI JNI work used it, and so on) and none that uses more than two levels.  All of the things where we've done whiteboard sketches of multiple levels end up needing at least one more level than whatever you initially think (you realise that you want one more for TLS, or one more for the hypervisor, and so on).

arichardson, 2025-05-28:
The basic support for levels was included in the base ISA in #611. Closing this issue. Please reopen or create a new issue if you see anything that is still missing.