riscv/riscv-cheri #572: Naming of the base extension

arichardson, 2025-03-29:
The current ARC placeholder suggestion is RV32CH/RV64CH and that's what I've used in #566.

However, multiple characters are a bit odd.
Looking at the misa letters that are marked as reserved in the priv spec I see G, J, K, L, O, R, T, W, Y, Z. 

I don't think G, J, and Z can be used so maybe one of the following could work:

 * K (for **K**apability): probably not since that is used for most **K**ryptography extensions,
 * L (for **L**onger pointers)
 * R (for **R**estricted pointers)
 * T (somewhat contrived: **T**ightly-bounded pointers)
 * Y (for capabilit**Y**)
 * W (for **W**ider pointers)

Can't think of any name with O, but if needed I'm sure we can come up with something.

jrtc27, 2025-04-02:
What about E vs I? How do we differentiate between 16 and 32 GPRs?

arichardson, 2025-04-02:
> What about E vs I? How do we differentiate between 16 and 32 GPRs?

That is a bit tricky - I guess we would have to go for a two char version: RV64YE?

tariqkurd-repo, 2025-04-23:
closing this one - see https://github.com/riscv/riscv-cheri/issues/580

closed 2025-04-23 by tariqkurd-repo