riscv/riscv-cheri #577: Does the "implicit ACPERM" in LC do the full instruction semantics?

tomaird, 2025-04-14:
The description for `LC` (amongst others) mentions doing an "implicit ACPERM" to clear certain permissions of the loaded capability. e.g.:

> If the authorizing capability does not grant LM-permission, and the tag of cd is 1 and cd is not sealed, then an implicit ACPERM clearing W-permission and LM-permission is performed to obtain the intermediate permissions on cd.

What is not clear is if "implicit ACPERM" means to just do the permission clearing part of the ACPERM semantics, or apply the entire instruction.

The ACPERM description has the following steps:

[image]

I think it's clear that steps 1-6 should be applied to clear the relevant permissions. But it's not clear if step 7 should also be applied as part of this process.

If so, this would mean that any loaded capability which is sealed or has non-zero reserved bits should have its tag cleared.

tariqkurd-repo, 2025-04-14:
If a sealed cap is loaded with write permission, which then can't be stripped as rule 7 isn't followed, then that probably isn't right?
Although the sealed cap can't be derefenced.
@arichardson @jrtc27 ? opinions?

tomaird, 2025-04-14:
Following internal discussions, we've agreed the "implicit ACPERM" should do the full instruction steps. Which is probably the most logical conclusion anyway.

However to make this clearer, I propose adding something like this to the `ACPERM` description:

```adoc
NOTE: Whenever an _implicit_ ACPERM is done (e.g. when clearing LM-permission), the full ACPERM instruction semantics are done, not just the permission clearing parts.
```

nwf, 2025-04-14:
In CHERIoT, `lc` does modulate permissions but is not a "full" `acperm`.  In particular, [`W` (AKA `SD`), `LM`, and `LG` permissions are "under seal" but the `GL`obal permission is not](https://github.com/CHERIoT-Platform/cheriot-sail/issues/14) and, as a consequence, while [`acperm` is allowed to clear `GL` on sealed capabilities if it clears no other bits](https://github.com/CHERIoT-Platform/cheriot-sail/issues/70) as if they had gone to memory and come back through a `LG`-absent authority, loading a sealed capability through a `LG`-bearing but `LM`-absent authority will leave the load result unaltered (that is, the capability loaded to the register file will be exactly the same as was in memory).

davidchisnall, 2025-04-14:
To add to @nwf's comment, the CHERIoT behaviour is the result of building software atop these rules.  In particular:

 - If this permission stripping removed W from sealed capabilities, a large number of unseal operations ends up needing an additional check that W has not been stripped.  This is a burden to programmers and the ability to strip W from sealed objects introduces subtle security issues if you omit it (for example, the ability to double-spend quota).
 - In contrast, it *is* useful to be able to strip global from sealed objects, so that you can delegate them for a single lexical scope.  The few places that depend on global not being stripped can address this with an interior pointer that has global permission if required because *load-global* remains under seal and is not stripped.

It would be unfortunate if the ratified spec got this wrong.

LawrenceEsswood, 2025-04-14:
@tomaird The first paragraph you quote has the condition

"and the tag of cd is 1 and cd is NOT SEALED" (emphasis mine)

It is therefore never a concern on what should happen on this implicit ACPERM on a sealed cap, because it would never apply to a sealed cap.

As @nwf says, apart from global, any fields "under seal" are protected from automatic squashing and if this is not the case (or detagging results) then there is a bug in the spec.

On reserved fields, you might be correct that we have not made a very deliberate choice on how they should interact with LC. If we conclude it doesn't really matter, because they never should be set, then I think its fine to clear them and so say all the ACPERM rules apply.

tomaird, 2025-04-14:
@LawrenceEsswood Perhaps this wasn't a good example. The following paragraph says:

> If the authorizing capability does not grant EL-permission, and the tag of cd is 1, then an implicit ACPERM restricting the Capability Level (CL) to the level of the authorizing capability is performed.

In this case the CL clearing is not dependent on whether the capability is sealed or not. So you're comment that "it's never a concern on what should happen on an implicit ACPERM for a sealed cap" is not exactly true.

LawrenceEsswood, 2025-04-14:
@tomaird Yeah, that is concerning. Something has been lost in trying to describe LM / EL as implicit ACPERMs. We either need to amend the text on EL to not imply that it would tag clear, or, change ACPERM to not tag clear if you are just changing level and nothing else on a sealed capability:

https://github.com/riscv/riscv-cheri/issues/513

I think it makes most logical sense to have consistency between how ACPERM behaves and how LC behaves. I would suggest that we fix this with issue 513, and continue to predicate LM on not applying to non-sealed things.

arichardson, 2025-04-16:
I also believe we should change ACPERM to permit level change on sealed caps, then we could simplify this whole section to just say "perform implicit ACPERM" without the if not sealed qualifier. The existing wording is very ambiguous so we could also fix this to remove "implicit ACPERM" from the level change while we wait for a conclusion on #513.

closed 2025-05-08 by arichardson