riscv/riscv-cheri #579: Make `cbld` optional?

nwf, 2025-04-22:
Hi, it's me with a CHERIoT-vs-Zcheri thing again, sorry.
TL;DR: can we add a note to the spec that `cbld` might be removed by Zcheri extensions, like they can opt out of `sentry`?

For more rationale, the CHERIoT ISA does not specify a `cbld`-alike, it's pretty unlikely that we'd use it even if we did:

- we have architectural (and virtual) sealing, which seems to supplant the use of `cbld`-as-`unseal`;[^cheriot-seals]

- we don't page/swap memory to external, tag-less storage;

- our toolchain does not emit capability-encoded bit patterns, but rather our loader explicitly performs capability derivation as required (driven by tables in the firmware image, `.cap_relocs` and others);

- even if we _did_ the above, CHERIoT's capability system is multi-rooted, so we'd want something more like the proposed `ccbuildcap` and `ccseal` semantics (see discussion in #298 and older notes); and

- the compiler doesn't, and probably won't ever, emit `cbld` as part of codegen unless directed to do so by a builtin intrinsic.

All that to say, it'd be quite nice if `cbld` could be _removed_ by an extension, even if it's provided by most implementations of Zcheri.  (This is somewhat thematically adjacent to the changes made to the Zcheri spec that let CHERIoT opt out of supporting `otype` 1 ambient sentries (and the `SENTRY` instruction), by using only other `otype` values, though I think it harder to find such a "cute" mechanism of justification.)

I suspect a note in the spec suffices?  Thoughts?

[^cheriot-seals]: Relative to Zcheri, CHERIoT (and its future Zcheri-based version) restores the notion of multiple architectural sealing types from the CTSRD CHERI ISA.  Atop this, among other things, the CHERIoT-RTOS adds support for _virtually_ sealed objects: a single (of the precious few) architectural `otype` is used to guard structures equipped with a header holding a much larger, software-managed `otype`.  CHERIoT software in practice makes heavy use of both the architectural and virtual sealing machinery.  The restriction that sealed objects are sealed under exactly one `otype` has not been a problem in practice, and, to the best of my knowledge, it's very rare even in "big CHERI" for the same span of address space to be held under more than one `otype` of seal at the same time or for a single object to change the single type of seal it's held under across time.  (But perhaps @LawrenceEsswood or @arichardson know of cases?)  Note that `cbld`-as-`unseal` has effectively the same restriction: the arena in which an object sealed under the `cbld`-as-`unseal` mechanism corresponds to the _single_ otype under which that object may be sealed.

nwf, 2025-09-03:
I am not sure why you are bringing those points up on _this_ issue.

In any case, my current proposal, in #716, leaves cbld in the base.  It doesn't hurt CHERIoT to have it, as it's just an optimization of something software could otherwise do.

jrtc27, 2025-09-03:
Sorry, I misread Tariq's comment as being current, not an ancient one. I skimmed the activity in this issue, saw it was closed, wondered why and saw the comment above, but of course that's from April. Don't mind me...