riscv/riscv-cheri #581: make a CHERIoT a new base architecture

tariqkurd-repo, 2025-04-23:
Krste says CHERIoT should be a new base architecture which inherits from RV32Y/RV64Y.

Therefore CHERIoT can choose what to inherit from RV32Y/RV64Y and make changes as required (e.g. the immediate format for AUIPC).

Therefore RV32Y/RV64Y will have a single fixed capability encoding which CHERIoT, or other capability based architectures in the future, will not need to inherit.

it also means that instructions such as `CBLD` which may not be needed for CHERIoT  do not need to be inherited.

davidchisnall, 2025-04-23:
I don't think that's an ideal approach, since it will complicate toolchains for little benefit.

We were converging on a place where CHERIoT could be a strict addition to RV32YE.  Stepping back from that seems like a bad idea.

SCI will vote 'no' on ratification of the CHERI spec if this is the route that ratification wants to go, unless we are able to ratify a CHERIoT extension at the same time as RV32YE.

nwf, 2025-04-23:
Moreover, I think this approach does not scale.  CHERIoT is surely not the only thing that wants to be "standard CHERI with some extra flare" (i.e., an extension to the RVnnY[IE] bases).  Real-world Big CHERI systems are going to want to add sealing, to fiddle with capability encoding and permissions and representability, to change instruction semantics, &c.  _Some_ of that is already acknowledged by things like Zcherihybrid and Zcherilevels being extensions to the base, and it seems a strange choice to make CHERIoT a fundamentally different thing.

davidchisnall, 2025-04-23:
And many of the things that CHERIoT wants (e.g. the reduced AUIPCC shift and a different capability format) are also important for some accelerator use cases.  So now they also won't be able to use RV{32,64}Y.

jrtc27, 2025-04-23:
That's not quite true though. CHERIoT doesn't really fit being a normal extension, and while there are some carve-outs in the current spec for it, they are quite weird. It uses an entirely different capability format. It removes instructions like sentry. It changes the encoding of auipc.

jrtc27, 2025-04-23:
I think the real point is that, if CHERIoT and any similar cases were to be deemed extensions to RVxY, that renders RVxY meaningless by itself. You can't generate code for it because you don't even know what auipcc does, and you don't know what the alignment/padding requirements are due to not knowing the capability format. You can't write a run-time for it because you don't know how to create (forward-edge) sentries. You can't write a debugger for it because you don't know how to decode the capability format. It becomes a worthless lowest-common denominator that always needs to be in combination with some additional information, which goes against how RISC-V base ISAs work - you can target all the existing ones directly.

davidchisnall, 2025-04-23:
> It uses an entirely different capability format. 

As will many other things.  Some accelerators will not want the current encoding, they will want something that can be packed into vectors with no our-of-bounds representation, stronger alignment rules, and fewer permissions.  And future versions of the base specification will need richer encodings because, if there's one thing that we've learned in the last 15 years of CHERI, it's that you don't think of all of the requirements until people have written (at the very least) hundreds of thousands of lines of code on your existing design, and then you end up with 

> It removes instructions like sentry. 

Technically, it keeps the instruction (it's a not a good idea as an instruction and should be a separate extension, but that's a separate discussion) it just doesn't provide an encoding for that otype and so it always gives an invalid capability.

And big CHERI will also need to remove this once it grows return sentries, because (unlike CSeal), it doesn't generalise to multiple kinds of sentry and things like stack unwinders are going to need to be able to generate return sentries and run-time linkers are going to need to be able to generate forward sentries.

> It changes the encoding of auipc

Which is something Y could do, since not doing it has zero benefit for any real-world code.  Even the monolithic Facebook back-end app doesn't have enough code to need the AUIPC shift and it's over an order of magnitude larger than the second-largest program I'm aware of.

davidchisnall, 2025-04-23:
> I think the real point is that, if CHERIoT and any similar cases were to be deemed extensions to RVxY, that renders RVxY meaningless by itself. You can't generate code for it because you don't even [...]

This doesn't make sense to me as an argument.  A lot of these also hold for accelerators based on RISC-V.  In most cases, you need to know the profile (or ad-hoc set of extensions) that are supported.  There are a lot of things you can't do with *just* RV32I (in particular, without atomics, you need to have some definitions for OS functions to support atomicity).  The point is that you can all of the things that are common to microcontrollers, 32-bit desktops, and accelerators are part of a common spec.  You don't ship something that is RV32I, multiple people ship things for which RV32I is a common subset and toolchains, debuggers, and so on can all reuse the code for the common subset in a principled way by simply referring to it.

ABIs are not ISAs.  Things like C++11's always-lock-free query depend on the set of atomics that you support as a baseline.  This is no different from the padding requirements arising from a capability encoding: the ABI needs to define a minimum set of things that the hardware must support.  

Profiles exist to define a set of things that must be included together to provide a useful platform.  There absolutely should be such a thing for a *NIX-capable CHERI.  There should be a profile for CHERIoT.  They should both have a common subset that is the Y base architecture.

tariqkurd-repo, 2025-04-23:
>They should both have a common subset that is the Y base architecture.

This is the most desirable outcome, I agree.
We have two high level goals here:

1. get the base Y architecture through ARC review
2. make it reusable for CHERIoT, and other CHERI architectures in the future

It's very true that we can't predict what people will want from CHERI in future, and need to be open to new extensions and features.

I think 2. will get easier when we've made more progress with 1. so ARC have a better understanding of CHERI and what we're trying to achieve.

I guess most of us will be at the summit in Paris in May along with at least some ARC members. This should be a good opportunity to work through some of the problems and improve the understanding of CHERI.

davidchisnall, 2025-04-23:
> I guess most of us will be at the summit in Paris in May

I'm not sure who the 'us' is there, I don't think any SCI or Microsoft CHERIoT folks are going to be there.

tariqkurd-repo, 2025-04-23:
Ok, unfortunate. Cambridge Uni are normally at such events in force and
I'll be there.

tariqkurd-repo, 2025-04-25:
I looked back through my notes from the meeting with Krste - he said that CHERIoT needs to be a new base as it redefines loads and stores (I didn't mention AUIPC but that's in the same category).

If we can't fully define the behaviour of an instruction, which we can't if the encoding format or permissions change, then it needs to be a new base.

So I think we need to find a way of getting that new base included as quickly and efficiently as possible.

I guess we can discuss at the TG meeting on tuesday.

nwf, 2025-04-25:
> If we can't fully define the behaviour of an instruction, which we can't if the encoding format or permissions change, then it needs to be a new base.

That perspective is... **deeply** worrying, to say the least, because it suggests that Zcherihybrid (which changes the semantics of load and store instructions to be indirected through the new DDC CSR) must also be a new base.  If we hadn't already known to add explicit hooks for sealing by extensions (decoupling the notion of "is sealed" from "is sentry"), that would also constitute a new base.  If we hadn't already known to add explicit hooks for Zcherilevels (in `lc` and `sc` and `acperm`, at least), that would also constitute a new base.  If we hadn't thought to add access-system-registers or recursive-load-mutable to the spec, those would be yet more bases.

Further, I think this perspective is not particularly compatible with the history of RISC-V.  It would suggest that the privileged spec's Hypervisor extension or Zfinx-and-friends (and maybe even Ztso) should have required separate bases.

I continue to think the correct target for CHERI standardization is a set of extensions, possibly to a new base architecture if ARC insists, _grouped into one or more platform(s)_, analogous to RV64G back in the day.  Toolchains should target _platforms_, and ideally the specification and the corresponding toolchain implementations thereof are reusable _across_ platforms, even if this costs a little bit of effort for parametricity (in, say, the padding required for representability or the exact set of named permission bits exposed to compilation targets).

tariqkurd-repo, 2025-04-25:
It is certainly true that hybrid mode alters load/store behaviour, and more
intrusively than just adding more exceptions or translating the address as
the checking changes. I think we need more clarification

arichardson, 2025-04-25:
hybrid alters loads/stores in the same way as a PMP (or an identity mapped MMU), but I guess it does add the M bit to the format. I am not convinced that you need a new base just because the bits you loaded happen to be interpreted differently (Zfinx as noted by @nwf is a good counterpoint here).

davidchisnall, 2025-04-26:
> hybrid alters loads/stores in the same way as a PMP

This is true if you consider them as applying to an I base, but on a Y base they change all loads and stores from taking a capability base to taking an integer base, and they redefine the capability checks as operating against DDC. That’s a bigger delta to Y than anything in CHERIoT.

jrtc27, 2025-04-26:
> This is true if you consider them as applying to an I base, but on a Y base they change all loads and stores from taking a capability base to taking an integer base, and they redefine the capability [...]

If you switch encoding mode then you switch base ISA to I/E, surely? The whole point of that is to be able to run legacy unmodified RV64I code, including binaries, for example.

closed 2025-09-05 by tariqkurd-repo