riscv/riscv-cheri #583: ASR-gating of CSRs?

nwf, 2025-04-23:
The current draft of the privileged specification is not particularly clear about ASR-gating of CSRs.

- "4.1. Machine-Level CSRs added or extended by Smcheri" says

  > [[asr_perm]](https://riscv.github.io/riscv-cheri/snapshot/privileged/#asr_perm) in the [[pcc]](https://riscv.github.io/riscv-cheri/snapshot/privileged/#pcc) is typically required for access.

  but none of its subsections detailing, respectively, `mtvecc`, `mscratch`, and `mepcc`, expand on that "typically".

- "4.2. Machine-Level CSRs modified by Smcheri" contains no text not in a subsection and "4.2.1. Machine Status Registers (`mstatus` and `mstatush`)" does not mention ASR-gating at all.  Neither does "4.2.2. Machine Cause Register (`mcause`)", but that is probably OK not to ASR-gate.

- "4.3. Supervisor-Level CSRs added or extended by Sscheri" contains the same "typically" as 4.1, but again the subsections do not detail further.

arichardson, 2025-04-24:
Thanks this is a good catch. How about we phrase this as "all privileged CSRs require ASR-permission unless stated otherwise"? I believe the only exception here would be that the TID registers are readable without ASR.

nwf, 2025-04-24:
I think that's right and agreeable to CHERIoT. :)

davidchisnall, 2025-04-24:
We also make mcycle and minstret readable (but not writeable) without ASR. I can imagine a later addition providing one or more CSRs to configure which CSRs (possibly from some subset) can be read or written without ASR. Some folks have concerns about exposing mcycle and minstret to normal compartments due to side channels and so we may want to restrict this to specific compartments and provide a privileged service that fast-path accesses mcycle if it is allowed to or a quantised version if not.