riscv/riscv-cheri #601: consider merging CHERI invalid address fault and CHERI bounds fault

tariqkurd-repo, 2025-05-07:
The unpriv spec is tricky to write as it can't assume the presence of virtual memory, giving this as the list of possible exceptions from load/stores:

**tag
sealed
permission
bounds**

When we get to the Vm section of unpriv we'll need to insert **invalid address** between permissions and bounds.

The point of the invalid address check is to shrink the size of the comparators for the bounds check to the valid range (e.g. 48-bit for Sv48) and also to allow upper address bits to be thrown away.

The point is that if the invalid address exception is required, then it means that the bounds *can't* be decoded as the address field has been modified.

Therefore I'm wondering if we should combine the invalid address and bounds exception codes.

Currently for bounds we say:

`At least one byte accessed is outside the authority capability bounds, or the capability has malformed bounds`

If we change this to

`At least one byte accessed is outside the authority capability bounds, or the capability bounds cannot be decoded`

Then I think we can merge the invalid address check into it and have a simpler defintion.

We won't need to add a new exception type as part of the VM chapter.

arichardson, 2025-05-07:
I can't think of a reason why this needs to be separate for software. In most cases I view the type of exception more as something useful for a debugger rather than something to actually act upon.

And if this is really needed for some use case, we can determine which of the case happened by looking at the capability.

Does anyone else have an opinion on this (maybe @jrtc27 @bsdjhb @brooksdavis)?

jrtc27, 2025-05-07:
It's rather odd to get a bounds fault when you have almighty bounds, yet this would do that

arichardson, 2025-05-08:
I agree this is odd, but does it actually *need* to be a separate code or is it just conceptually cleaner?

tariqkurd-repo, 2025-05-08:
It doesn't take the invalid address fault if the bounds are infinite (this screenshot is from 0.9.5), because in that case we know the bounds and so know what to do with the bounds check.

[image]

So I believe that merging the two exception codes does actually makes sense

bsdjhb, 2025-05-08:
To be clear, the case in question is when a capability with non-infinite bounds has been set to a non-canonical address?  (The upper N bits are not all the same?).  I suspect in most cases that address would be so far out of the capability's original bounds that it would be untagged.  Presumably an OS could screw this up if it doesn't clamp a user-only or kernel-only capability to clear the upper N bits, but that would be an OS bug rather than something userspace can trip over in the general-purpose OS case.  If my understanding is correct, then I do think this case is probably rare enough to not require its own exception. 

One might say that even if the address is canonical, if the current paging mode is SV48 and the bounds are not infinite but spill over into a non-canonical range, that a memory access should fail with a bounds exception?  I can't think of a reason why you would want to use such bounds legitimately, and that would also happen to handle non-canonical addresses as well.  Maybe that more general check is too expensive however?  (I'd be fine if attempting to set such bounds cleared the tag instead, but maybe it is better to only break during memory access to handle dynamic transitions between SV48 and SV57.  Experience on x86-64 is that different processes on the same system switch between 48-bit addresses and LA57.)

tariqkurd-repo, 2025-05-28:
Hi @bsdjhb sorry for the slow reply - the real intention behind the "invalid address check" is to reduce the size of the bounds comparators in the CPU. Previously (for an RV64 machine) the bounds calculations always had to be 64-bit, and so did the comparators which is quite expeneive if only SV39 is implemented.

Implementations are permitted to implement all 64-bits of address and 64-bit bounds comparators if they see fit, but if they decide to restrict the width of virtual addresses then we also allow them to restrict the width of the bounds comparators.

If a CPU is Sv39/Sv48 switchable then the CPU must implement (at least) 48-bit comparators.

So the point is that if the address is in the invalid space, and the CPU has not implemented all the necessary address bits to represent it, then there's no need to decode the bounds and do the comparison, just take the fault.

The `bounds fault` currently includes the `malformed bounds` case, in which case the bounds can't be decoded, so it seems reasonble to include this case too, and yes it should be extremely rare.

commit 8046b598 (2025-07-11): Rework exception reporting (#654)

Update with the ARC feedback allocating mcause values 32 onwards for CHERI faults.
Also merge invalid address and bounds fault and
stopp overloading CHERI PTE faults into normal RISC-V page faults.
We also reduce the number of faults to the minimum that needs to be differentiated: instruction, load, store plus the load and store PTE faults.
In theory we could merge the load and store access faults since any emulation will need to decode the instruction anyway, but keeping them separate is consistent with the base ISA.
This allows us to remove Xtval2.

Fixes:  [... cut]
closed 2025-07-11 by arichardson