riscv/riscv-cheri #606: CBLD is presently specified to construct arbitrary otypes.

nwf, 2025-05-13:
As of writing, [cbld_32bit.adoc](https://github.com/riscv/riscv-cheri/blob/1ffc7977db7ba014da46da2e521d1d777a8129a6/src/cheri/insns/cbld_32bit.adoc) and the [corresponding sail](https://github.com/CHERI-Alliance/sail-riscv/blob/4f1dae4d0d38348a83dd8f3d670264330c767a22/model/cheri/cheri_insts.sail#L523-L557) (which mostly relies on [`capIsSubset`](https://github.com/CHERI-Alliance/sail-riscv/blob/4f1dae4d0d38348a83dd8f3d670264330c767a22/model/cheri/cheri_cap_common.sail#L511-L531)) permit constructing a tagged capability that is both
- arbitrarily attenuated (or not) from a capability held in a register and
- **sealed under arbitrary otype**.

This is more or less uninteresting in the base spec, since the only sealed forms are the ambiently-available `SENTRY` (CT=1), but it is, as one might imagine, *wildly damaging* to extensions that would add back University-CHERI capability-mediated sealing mechanisms or use otypes for light-weight CFI (as we do on CHERIoT) or similar.

I'd ask for reconsideration of #579, but failing that, I think `cbld` should _set_ `cd.ct` to `0` regardless of the input `cs2.ct`.  Requiring that `cs2.ct` is zero is less pleasant: we don't have a way to reveal to software which bit(s) back the `.ct` field.  While we allowed [University-CHERI's `CBuildCap`](https://github.com/CTSRD-CHERI/sail-cheri-riscv/blob/74a9da1f78bdf50d263ffc98f2a587cf969b3e08/src/cheri_insts.sail#L768-L818) to also act as if (its spelling of) `SENTRY` had been applied, the Y spec would have to phrase the analogous act as "any ambiently-constructable, encoding-supported capability type" and I think that that is more awkward to specify, implement, and test than whatever benefits to software it might have would merit.

jrtc27, 2025-05-13:
When adding new kinds of sealed capabilities an extension is free to define how they work with CBLD. The current specification should not be taken to mean that every possible kind of sealed capability can be ambiently created.

nwf, 2025-05-22:
See also #199.

nwf, 2025-09-02:
See changes in #716.

jrtc27, 2025-09-04:
Sentries don't authorise the building of anything. Only unsealed capabilities do, and presently for any type, because there are only two and it's defined to be able to build both.

nwf, 2025-09-04:
Yes, sorry, I meant "every unsealed capability".  Edited to fix the omission.

closed 2025-09-20 by nwf