riscv/riscv-cheri #609: can CBLD be used to unseal cs2?

tariqkurd-repo, 2025-05-19:
CHERI v9 says as part of CBuildCap: `If cs2 is a sentry then cd is also sealed as a sentry`
CHERI-RISC-V 0.9.5 for CBLD says `Copy cs2 to cd and set cd.tag to 1 if...`

in neither case do they have the option of unsealing cs2

0.9.5 however says:

[image]

But this doesn't match either instruction definition, so is the note wrong?

arichardson, 2025-05-19:
I think the note is missing that you need to use SCHI first. A new instruction for unsealing is tracked in #199

nwf, 2025-05-19:
See also #606.

tariqkurd-repo, 2025-05-21:
I fixed the note as part of https://github.com/riscv/riscv-cheri/pull/610

closed 2025-05-21 by tariqkurd-repo