riscv/riscv-cheri #613: Naming of extended CSRs

arichardson, 2025-05-22:
Considering the ARC recommendation of using y for everything we could do the following:
`pc -> ypc`, `mtvec -> ymtvec`, etc.

Another alternative would be to not give a new name to the extended CSRs, and just use the existing names. The CSRRW instruction always shows whether you are accessing the extended or the XLEN CSR based on the other register operands (XLEN/CLEN destination/source). Do we even need a name for the extended one if assembly never needs to distinguish them?

The only remaining question is what to name new CSRs, namely DDC. Do we call it DDY or just keep DDC?

See #612

jrtc27, 2025-05-22:
We should be avoiding all use of Y outside of the extension name, not perpetuating its use in registers and instructions.

arichardson, 2025-05-23:
To be clear I prefer dropping the separate name for the extended registers. I don't see it adding any much value since we can just refer to the XLEN-subset of mtvec, etc. when talking about a CHERI system.

arichardson, 2025-05-23:
Maybe some the CHERIoT folks have opinions on naming here? @davidchisnall @nwf?

nwf, 2025-05-24:
In this sense, CHERIoT is much closer to University-CHERI-RISC-V than it is to Y/Zcheri.  We still have the separate "SCR" namespace and `cspecialrw` instruction with some CSRs aliasing the addresses of SCRs.

(Speaking only for the _software_ people, we'll happily follow the standardization effort in making CSRs up-to-capability-sized objects (with most of them having at most XLEN bits implemented); I _assume_ this change is acceptable to the hardware people, in that there are already capability-width data paths all over the chip, so making `csrrw` use one of those and sometimes connect more wires to some CSRs is fine (especially since we had to do that for `cspecialrw`).  Kunyan Liu may have more insight there.)

In terms of naming, we also were like University-CHERI-RISC-V: our augmented registers are named `pcc`, `mtcc`, `mtdc`, `mepcc`, `mscratchc`.  Certainly I would have no objection to dropping the `c` from those names (and renaming `mtdc` back to `mtvec` and our as-yet-unused `mtdc` to `mtid` if we want to pick up those semantics for the YXcheriot ISA).  All of this sounds like the kind of thing we can largely macro away in assembler to support both the CHERIoT-as-stands and the YXcheriot ISAs, and I can't imagine the compiler cares much either.  Suffixing with `c` didn't add much in terms of clarity, and I don't think prefixing with `y` would either.

I have no preference for DDC vs DDY.

More broadly, I think I'd even prefer getting rid of the `c` prefix on the GPRs (and not using `y` either).  It's vaguely annoying to have to pass the "integer" register name to assembler macros that then use `c\arg` when in use with capability instructions; one has to remember that the macros behave differently from instruction mnemonics.  If it's essential to differentiate between XLEN and 2*XLEN transfers (say, with `csrrw`?), I'd rather it be done like the load/store instructions: the _mnemonic_ should indicate the transfer size.  (FWIW, all of the documentation I wrote for the CHERIoT RTOS switcher, which is certainly the largest and most documented RISC-V assembler program I've ever maintained, consistently uses the integer names, because, in part, one always has to _search_ for the integer name to be sure to find the nearest uses of the register.)

tariqkurd-repo, 2025-06-19:
[image]

tariqkurd-repo, 2025-06-19:
[image]

tariqkurd-repo, 2025-07-15:
A further suggestion from ARC is just to stick with the normal name `PC`, `MTVEC` etc.