riscv/riscv-cheri #616: EF = 1 Overspecified

jonwoodruff, 2025-05-27:
The 32-bit format can use EF=1 up to 2^(MW-1) rather than 2^(MW-2), which is listed in the spec.

(To find the relevant line, search "The exponent is 0").

Actually, the entire phrase: "for regions less than 2MW-2 bytes long" is probably not necessary, since a literal reading of the line also implies that, when EF=1, the exponent would be non-zero for regions larger than 2^(MW-2).  Actually, when EF=1, E=0. Depending on L8 existing, this state might be used for regions less than 2^(MW-2) (no L8) or 2^(MW-1) (yay, have L8).

jonwoodruff, 2025-05-27:
For completeness, this issue was noted by @rmn30.

tariqkurd-repo, 2025-07-11:
I thought we'd merged a PR for this but don't seem to be able to find it. @jonwoodruff ?

arichardson, 2025-07-29:
@jonwoodruff is this still an issue?

tariqkurd-repo, 2025-09-05:
closing as I believe it's addressed by https://github.com/riscv/riscv-cheri/pull/641