riscv/riscv-cheri #621: Need for CLoadTags?

jrtc27, 2025-06-04:
CLoadTags has two uses:

1. Accelerate scanning for capabilities in pages
2. Accelerate saving the tag bits when swapping a page out

One or both of those may well motivate reintroducing the instruction from ISAv9.

jrtc27, 2025-06-06:
PMAs could let systems give an access fault for uncached regions I suppose. Much of that will be for memory that can't hold tags anyway.

> I agree that it is needed for performance, but I think it's too soon.

So we're just going to put out a spec that doesn't perform adequately? That's not going to end well, the net result will be people observing poor performance and writing off CHERI as a result. That would be disastrous for the technology.

tariqkurd-repo, 2025-06-06:
The intention is for the base arch to be a minimal viable set of instructions.
I think there are lot of things we can add.
However - if memory sweeping is really so poor without CLoadTags then may be we need to do something.
I don't think you'd want to mandate it though, a small embedded core with a cache wouldn't need to suport it for example, and probably wouldn't want the complexity.
there's a more compelling argument to mandate it for RV64, which is normally application cores only.

davidchisnall, 2025-06-06:
CLoadTags has a lot of impact on memory subsystem / cache-coherency protocol design.  In the server chips that we were working on, we didn't plan to use it and had some alternative mechanisms.  In CHERIoT, we don't use it (sweeping is not done on the CPU except in the Sail model, and even there it relies on the load filter not on explicitly inspecting tags).  Oh, it also has some very complex interactions with spectre mitigations if you want to make it fast and also make everything else resistant to transient execution attacks.

I don't object to it being in an early ratified Yloadtags extension or similar, but I would not want to implement it on microcontrollers or on server-class chips.  There's probably a middle range of microarchitectures where it makes sense, but a much larger range at the top and bottom of the scale where it doesn't.

tariqkurd-repo, 2025-06-06:
What you really want is a way to access the tags and _not_ the data in a cache coherent way.

I can imagine CLoadTags being used to extract tags from a line which is already held in the D$ and return them more efficiently than continual LC+GCTAG sequences, combining the resulting tags.

But again, what you really want is to query the L1 D$, L2$, L3$, tag cache to simply find the presence of tags, and that mechanism doesn't exist (as far as I know anyway). You don't really want to pull the data all the way into the L1 D$ to examine the tags.

davidchisnall, 2025-06-06:
At a minimum, you need an extra state in your cache coherency protocol that says that a line contains tags but not data and a bus message to fetch the tags.  And, if you want your implementation to not be completely terrible, you need to fetch tags + data if they are already together in another cache line somewhere else in the system.

nwf, 2025-06-07:
FWIW, I'm happy for CLoadTags to be an extension atop the Y base:
- as said, CHERIoT doesn't use it,
- doing it well places demands on the cache fabric and/or tag controller, and
- for all that there _are_ sensible cache-coherent semantics for it, those aren't necessarily actually the semantics a Reloaded-style revoker actually wants! [^0]

As far as the performance of revocation specifically, I think that hasn't been a strong driver of Y's design elsewhere (the page table semantics in the draft are, AFAICT, going to pose a challenge; see Alfredo's excellent deep dive into that).

[^0] Specifically, Reloaded might benefit from seeing a view of tags that need not be coherent so long as it is correct as of the current epoch's start.  Subsequent 0->1 tag transitions are safe to miss, as the written cap must have been previously scanned, and 1->0 transitions will be caught (and unaltered) in the sweep's probe (or coherent, atomic update).